FreeCAD/FreeCAD #11974: [Problem] disappearing lines in sketcher

greggjaskiewicz, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Issue in the new weekly snapshot. 
I'm trying to work through the exercise (see attached screenshot), and adding lines for the second bits "the wings", just causes some of them to randomly disappear. 
Very odd. 
See attached video

https://github.com/FreeCAD/FreeCAD/assets/553583/f0ff98d2-8b33-43b2-b364-3c5388b8c18f
[ex90.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13929274/ex90.FCStd.zip)
[image]

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35570 (Git)
Build type: Release
Branch: main
Hash: 3ba0c3d7958dad891e125e065a41d95faac47c6e
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-01-13:
It seems that it wants to auto constrain point on line and tangent at the same time, where tangent only would be enough.

luzpaz, 2025-06-24:
@johnocfii any chance you can test this on macOS ?

JohnOCFII, 2025-06-24:
Using a fresh build from today, I did not see this issue.  The original issue was posted 6 months ago, so certainly there have been many changes in Sketcher since that time.

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Unknown
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.6.1
  * Telemetry 1.0.1
```

[image]

closed 2025-06-25 by maxwxyz