FreeCAD/FreeCAD #11975: Symmetry constraint broken for two points around line

dholstei, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Symmetry constraint is broken when the bottom points of the upper rectangle are made symmetric to the vertical center line.  The left bottom point of the rectangle is constrained to be on the line and 1 inch from the left
[image]

### Full version info

```shell
[code]
OS: openSUSE Leap 15.2 (KDE//usr/share/xsessions/plasma5)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-01-13:
It looks like your sketch is over-constrained. Does it say something in the task dialog? Could you share the file?

There is also a newer stable version of FreeCAD available, 0.21.2 as you are using 0.20.2
https://www.freecad.org/downloads.php

dholstei, 2024-01-14:
It had **only one DOF** before applying the symmetry constraint, the two right points and line could be dragged around.  When I applied only the symmetry condition is what you see.

It's a **bona fide** bug, I don't want to proceed and install a newer version on the **_possibility_** it'll be fixed.

I couldn't attach the FreeCAD file but it's easy to reproduce.  Create  a rectangle, then another rectangle constrained to the top line, set the distance from the left, then apply the symmetry constraint.

maxwxyz, 2024-01-14:
This is not a bug, you just over-constrained your sketch. The symmetry constrain on two points is in redundant with the horizontal constraint. Remove the horizontal constraint and it should be good. Look out for messages in the status bar or report view.

Check if your version of FreeCAD already has the `Auto remove redundants` preference. If checked, FreeCAD handles these cases and removes the redundant constraints:
[image]

Alternatively you can disable all auto constraints and constrain the sketch manually.

Off topic: I think bugfixes won't be merged to old versions of FreeCAD, at least not 0.20.

Roy-043, 2024-01-14:
@dholstei As explained this is not a bug, but expected behavior. I am closing this as a user error. Please first consult the [FreeCAD forum](https://forum.freecad.org/) before creating future GitHub issues. Thanks.

closed 2024-01-14 by Roy-043

dholstei, 2024-01-14:
I understand the logic and how "Auto remove redundants" works.  As a **user**, I expected it to add **only** the distance from the line since it was already perpendicular to the line.

maxwxyz, 2024-01-14:
It has nothing to do with the distance constraint and in this case there is no perpendicular constraint at all. The auto constraint "point on line" was added when you drew the second rectangle and *snapped* it to the first line. The "symmetry" is then the redundant one you created manually on top.
So either disable auto constraints or check the box for auto remove redundants if the latter is the behavior you want.

dholstei, 2024-01-14:
I did check that box and it works as you described.  And I understand what happened with the snap-to-line, that's what I had intended.

I was referring to the perpendicularity the symmetry constraint **added**, it makes sense to me now, but my goal was only to add a distance symmetry.  The over-constrained message makes no sense,  the default preference should be auto-remove.

I guess what I read about the learning curve is no exaggeration.