FreeCAD/FreeCAD #11976: [Feature Request] Sketcher: Add helpers to arcs for angle and distance constraints rendering

maxwxyz, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Following https://github.com/FreeCAD/FreeCAD/pull/11507 with respective PR https://github.com/FreeCAD/FreeCAD/pull/11926 to add helpers in Sketcher WB for radius and line angle, additional helpers for arcs would be useful:
- arc to other geometry: distance
- arc: angle

current implementation:
[image]

New issue from:
https://github.com/FreeCAD/FreeCAD/pull/11926#issuecomment-1890750085

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-14:
Doesn't [Sketcher_ArcOverlay ](https://wiki.freecad.org/Sketcher_ArcOverlay) already answer part of this feature request?

maxwxyz, 2024-01-14:
only some part

closed 2024-01-22 by chennes