FreeCAD/FreeCAD #11977: getGlobalPosition does not work for linked objects.

KeithSloan, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For normal FC objects one can get their global Placement by calling getGlobalPlacement but for linked Objects this does not work.

If a user application needs to find/work with Global Placements then it is probably more important that this function is available for a link than an object as there can be a number of linked objects all pointing to the same object

### Full version info

```shell
[code]
OS: macOS 12.6.8
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ImportNURBS 1.1.0
  * ImportExport_3DM 0.1.0
  * DynamicData 2.53.0
  * Pyramids-and-Polyhedrons
  * occi-freecad-plugin 0.1.0
  * GDML 2.0.0
  * Design456 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Curves 0.6.14
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-01-20:
+1, I add to reimplement a function to get global placement in assembly workbench because of that. See in https://github.com/FreeCAD/FreeCAD/pull/10764

WandererFan, 2024-01-20:
Another +1.  There is a bunch of code in TechDraw for getting the located shape of a link.  Would be much cleaner if it could just call getGlobalPlacement(linkObject).

KeithSloan, 2024-01-20:
> Another +1. There is a bunch of code in TechDraw for getting the located shape of a link. Would be much cleaner if it could just call getGlobalPlacement(linkObject).

In TechDraw is it in C++ code or Python? A C++ implementation would be preferable.

WandererFan, 2024-01-20:
Mod/TechDraw/App/ShapeExtractor.cpp

ShapeExtractor::getXShapes(const App::Link* xLink)

KeithSloan, 2024-01-20:
> Mod/TechDraw/App/ShapeExtractor.cpp
> 
> ShapeExtractor::getXShapes(const App::Link* xLink)

That is good, just need code added to getGlobalPlacement and TechDraw updated to use if FC is a version that supports.

KeithSloan, 2024-01-21:
Just to add a couple of related Forum posts

- https://forum.freecad.org/viewtopic.php?p=390274&hilit=getGlobalPlacement#p390274
- https://forum.freecad.org/viewtopic.php?t=84327&start=10

Hopefully the TechDraw code is a complete solution.

luzpaz, 2024-03-15:
What's the status of this ?

PaddleStroke, 2024-03-15:
FIY in assembly I have two versions of this getGlobalPlacement, one in python (UtilsAssembly.py) and one in cpp (AssemblyObject.cpp). They are translation of each other.
It seems to work in all cases as far as I know.

KeithSloan, 2024-03-15:
I assume these are in 0.22

Trying with Ondsel which I assume is also 0.22 I tried

`
from Assembly import UtilsAssembly
`

and replacing original call

`
#s = Part.getShape(t[1], mat=t[1].getGlobalPlacement().Matrix, transform=False)

s = Part.getShape(t[1], mat=t[1].UtilsAssembly.getGlobalPlacement.Matrix, transform=False)
`

But get error
`
18:33:58  pyException: Traceback (most recent call last):
  File "/Users/keithsloan/Library/Application Support/Ondsel/Macro/ProcessDir_BrepShell_MMHgetGlobalLinks.FCMacro", line 240, in <module>
    base = dirBase_class()
  File "/Users/keithsloan/Library/Application Support/Ondsel/Macro/ProcessDir_BrepShell_MMHgetGlobalLinks.FCMacro", line 48, in __init__
    self.processGDMLfile(fn)
  File "/Users/keithsloan/Library/Application Support/Ondsel/Macro/ProcessDir_BrepShell_MMHgetGlobalLinks.FCMacro", line 181, in processGDMLfile
    from Assembly import UtilsAssembly
<class 'ImportError'>: cannot import name 'UtilsAssembly' from 'Assembly' (/Applications/Ondsel ES.app/Contents/Resources/Mod/Assembly/Assembly/__init__.py)
`

What is the correct way to use the version in the Assembly workbench?

PaddleStroke, 2024-03-16:
`UtilsAssembly.getGlobalPlacement(obj, container)`

It returns the placement of obj as seen from container.
Container can be the obj itself, a App::Part, a PartDesign::Body, a App::Link. It is the most exterior container. The reason why we need it is for external objects.
Say you have this : 
```
doc1.part.cube
doc2.assembly.LinkToPart.cube
doc2.assembly.LinkToPart2.cube
```

and you want the placement of Cube. There are multiple answers if you only consider cube. So that's where container comes in. If you want the placement of cube as seen from linkToPart you do : 
UtilsAssembly.getGlobalPlacement(cube, LinkToPart2)

KeithSloan, 2024-03-18:
> `UtilsAssembly.getGlobalPlacement(obj, container)`
>
> It returns the placement of obj as seen from container. Container can be the obj itself, a App::Part, a PartDesign::Body, a App::Link. It is the [...]

By most exterior container, does it mean the rootObj for a Linked object?

i.e.
If I have something like

```
Object Obj

App::Part rootObj
        App::Part Part
              Linked Object (obj)
  ```            
 Is there a function to get rootObj for an Obj

KeithSloan, 2024-03-18:
I am trying to use in a Macro that creates a Brep of the Outer Shell of objects.
To work out the outer shell I have to test if an Object is totally inside another.
As the Objects can be in other Parts etc I need to make sure all Shapes have their GlobalPlacements
I have coded as follows but it appears I am not picking up the Global Placements

`
                # Needs FreeCAD 0.22 or later
		print(f"UtilsAssembly - need FreeCAD 0.22 or later")
		import UtilsAssembly
		
		for t in inputList:
			#s = Part.getShape(t[1], mat=t[1].getGlobalPlacement().Matrix, transform=False)
                          o = t[1]
                          if hasattr(o,"LinkedObject"):
                                gpMat = UtilsAssembly.getGlobalPlacement(o.LinkedObject, o).Matrix
                          else:
                                 gpMat = UtilsAssembly.getGlobalPlacement(o).Matrix
                          s = Part.getShape(o, mat=gpMat, transform=False) 
                          shapeNames[s] = o.Name
                          shapes.append(s)
		testList = self.sortObjList(shapes)
`

KeithSloan, 2024-03-18:
Test File 
[testFile.FCStd.txt](https://github.com/FreeCAD/FreeCAD/files/14643202/testFile.FCStd.txt)

KeithSloan, 2024-03-18:
Okay I think I have found a problem which I need to resolve first.

PaddleStroke, 2024-03-19:
> What if I have something like
> 
> PartPythonFeature obj
> 
> App::Part rootObj
>         App::Part Part
>               Linked Object (obj)
> Can I just use
> 
> UtilsAssembly.getGlobalPlacement(obj, linkedObj)

Yes this should be ok.

kadet1090, 2024-12-11:
https://github.com/FreeCAD/FreeCAD/pull/18437 - will this PR help here?

maxwxyz, 2024-12-25:
is getGlobalPLacement deprecated? If so, I'd close this issue in favor of https://github.com/FreeCAD/FreeCAD/issues/18664

PaddleStroke, 2024-12-25:
Yes this can be closed I think. The old function is deprecated.

closed 2024-12-25 by maxwxyz