FreeCAD/FreeCAD #11978: [Problem] crash when creating ellipse

greggjaskiewicz, 2024-01-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

drawing an ellipse by centre, major radius point has caused a random
 crash in sketcher. 
I have not seen this crash before and I wasn't able to reproduce it again to record a video.
[crash1.txt](https://github.com/FreeCAD/FreeCAD/files/13929600/crash1.txt)

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35570 (Git)
Build type: Release
Branch: main
Hash: 3ba0c3d7958dad891e125e065a41d95faac47c6e
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-01-14:
Is this related to your issue https://github.com/FreeCAD/FreeCAD/issues/11888

Roy-043, 2024-01-14:
If it is not reproducible it may be better to close this one.

greggjaskiewicz, 2024-01-14:
It does happen but I can’t reproduce it each time. Different thing.You got a stack trace there !

luzpaz, 2024-08-25:
@greggjaskiewicz have you encountered the bug again ? 

Note: I wonder if this is macOS ARM specific ?

luzpaz, 2024-09-07:
@greggjaskiewicz what is different between this ticket and #11888 specifically ?

benj5378, 2024-09-12:
@luzpaz, I got @JohnOCFII to test on MacOS, he couldn't reproduce it. If we can't recreate it, then it's impossible to debug. With no new info for 6 months, I think we should close this and #11888 for now.

luzpaz, 2024-09-12:
Closing as no response from OP or reproducibility.

luzpaz, 2024-09-12:
looks like @greggjaskiewicz is active. If you can reproduce please re-open.

closed 2024-09-12 by luzpaz

benj5378, 2024-09-12:
1) Reproduction steps - that some dev can reproduce
2) Example file