FreeCAD/FreeCAD #11979: [Problem] Remembered window position when changing multiple screen setup: dialogs are unable to reach

maxwxyz, 2024-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working in FreeCAD with a three monitor setup on Windows and then changing to a two monitor setup, some FreeCAD dialogs and windows are out of reach. They are rendered at the missing third monitor. (maybe related to https://github.com/FreeCAD/FreeCAD/issues/11142 )
This is e.g. the preferences window. I can get the preference out of the void with WIN + Shift + -> but not all windows and dialogs are possible to reach.
For example the `Edit parameters...` dialog is unable to reach at all, on my Laptop when I do not have three monitors plugged in. When opening the dialog, it is active on a missing screen and I cannot see, interact or close the dialog.

Then I have to open the Task Manager and end the FreeCAD process to work again. All unsaved changes are lost.

https://github.com/FreeCAD/FreeCAD/issues/11142#issuecomment-1784642910

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.25.1
  * CurvedShapes 1.0.5
  * Curves 0.6.21
  * Defeaturing 1.2.1
  * fasteners 0.5.0
  * FEMbyGEN 2.1.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.15
  * woodworking 0.21.1.33694
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-01-26:
I can reproduce this, I'm not sure but we might need to add checks for available screens for each window individually

chennes, 2025-06-30:
I'm curious if this is fixed in Qt6 -- I don't see any calls to `saveGeometry()` or `restoreGeometry()` in our codebase at all, so we aren't "remembering" the window positions as far as I can tell.