FreeCAD/FreeCAD #11984: [Problem] Stylesheets are not coherent in code and looks and are outdated.

MisterMakerNL, 2024-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad now has 

1. "Dark"
2. "Darker"
3. "Dark modern"
4. "Dark behave"
5. "Dark contrast"
6. "ProDark"
7. "Light"
8. "Light modern"

8 Stylesheets, but 3 of them are the same and only have different colors.
2 of the 8 are slightly changed from the 3, these are the modern version. These are there to try and make a more modern style. Which isn't going that fast. But they have the same code only the tabs and buttons style are different.
Basically the 5 are the same style in code and looks.
Then we have the 3 others:
Which are:
1. "Dark behave"
2. "Dark contrast"
3. "ProDark"
They all have different code style and look very different.
They are also not up-to-date the custom color setting and the overlay system have not been checked and there are very minor bugs around.
Example tiny bugs:
These buttons in Prodark
[image]
No highlight in the pull-down:
[image]
Overlay in Dark behave same with contrast.
[image]
All the overlay icons aren't good too.
So not really unusable but also not great.

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/pull/11982

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

RohanMishra315, 2024-01-15:
Hey @Roy-043 @MisterMakerNL I would love to look into this ?

Syres916, 2024-01-19:
See https://github.com/FreeCAD/FreeCAD/pull/12039

chennes, 2024-01-20:
I'm sure all of the stylesheets have some bugs in them, and just like every other piece of code, there are parts that are more actively maintained than others. Let's be careful to separate technical problems (text cut off, buttons the wrong size, etc.) with aesthetic considerations (e.g. "looks outdated"). We don't blindly remove sections of code that no one is working on unless there is a good reason for it, so a discussion about winnowing the number of themes provided by default is its own Issue and should be discussed separately of technical or aesthetic considerations of the themes themselves.

MisterMakerNL, 2024-01-20:
@chennes It's not about personal choice on looks. More about providing a more polished/uniformly look and keeping it like that over time.
It's also not to withheld options, they definitely should be placed in the addon browser if people are using them.
I just hope there will be someone to maintain them then.

I said it was outdated in looks because of the missing overlay stylesheets mostly.

I do think I should undo behave dark since someone stepped up to maintain it.

maxwxyz, 2024-04-15:
To add from the FreeCAD Day 2024 Complaint Session:
**When you change the theme it should change everything in contrast with that theme. If you use a light theme then you need to change other colours to make it work.**
Discussion:
When using a light theme, in the sketcher I always have to change, for example, the line color because it’s white, it’s not visible. So it would be nice to have a like a style-aware theme, if you change it, change it to something light, it is actually visible, so you don’t have to mess around with additional color settings.
When you change the theme it should also change everything that should be in contrast with that theme to actually be in contrast and legible.
Also when I just changing the background (for example to white) the rest everything is fine and then nothing else is visible.
If you use a light theme then you need to change the color of many things to make it work.

MisterMakerNL, 2024-04-15:
I hope we fix the line color issue being hardcoded into the Freecad file with the new Material overhaul!!! Since I think the sketch line color was similar to adding a part color.

furgo16, 2024-05-11:
Given the direction the Start page is already taking (choice of 3 themes: Classic, Dark, Light), and given that the 1.0 release is shaping up, what are the blockers in this discussion?

[image]

MisterMakerNL, 2024-05-11:
I made a dark style that is the same as the no-stylesheet light-style:
https://github.com/FreeCAD/FreeCAD/pull/13772
This way we can have 3 settings:
Auto (follow system)/dark/light

Reqrefusion, 2024-06-07:
I am one of those who think this problem should be solved. The steps should be like this:

- Themes that are not actively maintained should definitely be removed. This seems like the thing to do anyway, but I didn't see any PR so I felt the need to write this.

- For the themes being maintained, only one of them should have dark and light in their names. I think this solves most of the confusion. And it would be good to change bad names too. For example:
Dark Behave-Blue Night @Syres916 
It is a much more elegant and beautiful-sounding name. Of course, this is just a suggestion, better names can be found.

maxwxyz, 2024-06-30:
@MisterMakerNL can this be closed now?

MisterMakerNL, 2024-06-30:
Dark behave discussion is going to be moved too an addon so yes!

closed 2024-06-30 by MisterMakerNL