FreeCAD/FreeCAD #11988: [Problem] Part transparency of parts ignored in views

furgo16, 2024-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a window assembly where the glass panels are made transparent by setting the Object style > Transparency property to 85, which works as expected in the 3D view.

However, when I add that view to a TechDraw page, the transparency value seems to be ignored and the glass appears as opaque:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35578 (Git) Snap 895
Build type: Release
Branch: main
Hash: ce1a43ed2cdc3e7688feecfedd49df26e4234043
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * BIM 2021.12.0
  * ExplodedAssembly
  * Assembly4 0.50.6
```

### Subproject(s) affected?

Techdraw

### Anything else?

[File to reproduce](https://www.dropbox.com/scl/fi/kgf77zkrcos0hdp1q29ho/Exploded_transparency.FCStd?rlkey=a35tgau31k91coa8cam4xpe4x&dl=1)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-15:
TechDraw does not consider the View properties of an object. As a workaround you can select the edges of the transparent object, create a [PartDesign_SubShapeBinder](PartDesign_SubShapeBinder) and use that for the view.

Roy-043, 2024-01-16:
I don't think the current behavior should change. Transparency is also use to help with modeling. It need not mean the object is actually transparent.

@WandererFan Should this be closed?

furgo16, 2024-01-18:
Thanks for looking into this and for offering a workaround, really appreciated.

> I don't think the current behavior should change. Transparency is also use to help with modeling.

At the point of modeling though, if a non-transparent object is made transparent, would that not be a temporary measure to help modeling and thus not necessarily need to be transferred to the drawing? That is, before a PDF with the drawing is exported, I would imagine the designer would set transparency of non-transparent objects back to 0.

Or is it customary to not represent transparency in 2D drawings and for glass to be shown as opaque?

Roy-043, 2024-01-18:
> Or is it customary to not represent transparency in 2D drawings and for glass to be shown as opaque?

I don't know.

luzpaz, 2024-02-12:
@WandererFan would you mind weighing-in on this issue ?

closed 2024-02-14 by WandererFan