FreeCAD/FreeCAD #11989: [Problem] Segfault on "Move Object to Other Body"

bsloane1650, 2024-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are 2 symptoms observed in the attached project file. The more severe one is a segfault. To reproduce:

- rename buggyIntersection.FCStd.zip to buggyIntersection.FCStd (.zip extension added to make github happy)
- Launch freecad
- Open the attached buggyIntersection project
- Enter Part Design
- In the Model view, double click "Top" to make it active
- Right click AdditiveLoft001 (under Top) and select "Move Object to Other Body"

The first time performing the above actions results and freecad quickly terminating, with the attached stacktrace being printed to the terminal. Subsequent attempts to reproduce the issue result in freecad hanging indefinitely with nothing printed to the terminal. Switching versions from 0.21.2R33771 to 0.22.0devR35570 caused the subsequent attempt to exit with a stack trace, however further attempts on 0.22.0devR35570 continued to hang.

This issue was encountered while trying to isolate a bug related to binary intersections, which is still observable in the attached project. The "Common" element should appear as a grid of square based pyramids (which the outer perimeter of pyramids cut off in the middle). Instead, it shows only an arbitrary subset of the pyramids. This subset can be changed by: Marking "AdditivieLoft001" for recompute. Recomputing "Common". On 0.22.0devR35570, performing this recompute operation generates the message:
   `21:54:51  Part::FeaturePython: Link(s) to object(s) 'AdditiveLoft001' go out of the allowed scope 'Array001'. Instead, the linked object(s) reside within 'Body002 Body002'.`

The attached project file began in freecad 0.19.2 (as packaged by Linux Mint 21.2). Once the initial problem was observed, I tried began attempting to isolate it in both 0.19.2 and 0.21.2 (from the freecad website). The crash was observed in bot 0.21.2 and 0.22.0dev. All versions showed the same behaviour regarding the buggy intersection.

### Full version info

```shell
OS: Linux Mint 21.2 (xmonad)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35570 (Git) AppImage
Build type: Release
Branch: main
Hash: 3ba0c3d7958dad891e125e065a41d95faac47c6e
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Anything else?

[terminalOutput_22.txt](https://github.com/FreeCAD/FreeCAD/files/13934298/terminalOutput_22.txt)
[buggyIntersection.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13934299/buggyIntersection.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bsloane1650, 2024-01-15:
After a bit more debugging, I do have a workaround for the intersections issue. The attached project is taking the intersection of a part (Body003) and a rectangular array (TopSpikes). It works as expected if I first create a binary union of TopSpikes (and nothing else), then take the intersection of Body003 and that union. After making that change, the above segfault still occurs as described.

xtemp09, 2024-01-15:
If somebody feels like digging this, applied is the backtrace.

<details><summary>Backtrace</summary>

```cpp
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x5555fd57d000)
    frame #0: 0x00007fff5615749e PartDesignGui.so`PartDesignGui::collectMovableDependencies(features=size=1) at Utils.cpp:526:65
   523                  unique_objs.insert(static_cast<App::DocumentObject*>(sk));
   524              }
   525              if (auto prop = static_cast<App::PropertyLinkList*>(prim->getPropertyByName("Sections"))) {
-> 526                  for (App::DocumentObject* obj : prop->getValues()) {
   527                      unique_objs.insert(obj);
   528                  }
   529              }
```

```cpp
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x5555fd57d000)
  * frame #0: 0x00007fff5615749e PartDesignGui.so`PartDesignGui::collectMovableDependencies(features=size=1) at Utils.cpp:526:65
    frame #1: 0x00007fff56144925 PartDesignGui.so`CmdPartDesignMoveFeature::activated(this=0x0000555559432ed0, iMsg=<unavailable>) at CommandBody.cpp:691:104
    frame #2: 0x00007ffff6cdd788 libFreeCADGui.so`Gui::Command::_invoke(this=0x0000555559432ed0, id=0, disablelog=<unavailable>) at Command.cpp:444:26
    frame #3: 0x00007ffff6cddf48 libFreeCADGui.so`Gui::Command::invoke(this=0x0000555559432ed0, i=0, trigger=TriggerAction) at Command.cpp:408:12
    frame #4: 0x00007ffff6cbd6bd libFreeCADGui.so`Gui::Action::onActivated(this=<unavailable>) at Action.cpp:112:22
    frame #5: 0x00007ffff6cca959 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:152:20
    frame #6: 0x00007ffff6cca942 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] void QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:185:95
    frame #7: 0x00007ffff6cca942 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(which=<unavailable>, this_=<unavailable>, r=<unavailable>, a=<unavailable>, ret=<unavailable>) at qobjectdefs_impl.h:418:49
    frame #8: 0x00007ffff42e33db libQt5Core.so.5`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(a=0x00007fffffffa180, r=0x0000555555cb6d10, this=0x0000555558c83250) at qobjectdefs_impl.h:398:57
    frame #9: 0x00007ffff42e33bf libQt5Core.so.5`void doActivate<false>(sender=0x000055555a09fe00, signal_index=4, argv=0x00007fffffffa180) at qobject.cpp:3925:30
    frame #10: 0x00007ffff42da3ab libQt5Core.so.5`QMetaObject::activate(sender=<unavailable>, m=0x000000000071a2c0, local_signal_index=1, argv=0x00007fffffffa180) at qobject.cpp:3985:26
    frame #11: 0x00007ffff4ed96c8 libQt5Widgets.so.5`QAction::triggered(this=<unavailable>, _t1=<unavailable>) at moc_qaction.cpp:376:26
    frame #12: 0x00007ffff4edcb8b libQt5Widgets.so.5`QAction::activate(this=0x000055555a09fe00, event=Trigger) at qaction.cpp:1161:27
    frame #13: 0x00007ffff509d912 libQt5Widgets.so.5`QMenuPrivate::activateCausedStack(this=0x0000555559d13350, causedStack=0x00007fffffffa298, action=0x000055555a09fe00, action_e=Trigger, self=true) at qmenu.cpp:1384:25
    frame #14: 0x00007ffff50a548e libQt5Widgets.so.5`QMenuPrivate::activateAction(this=0x0000555559d13350, action=0x000055555a09fe00, action_e=Trigger, self=true) at qmenu.cpp:1461:24
    frame #15: 0x00007ffff50a6754 libQt5Widgets.so.5`QMenu::mouseReleaseEvent(this=<unavailable>, e=0x00007fffffffab40) at qmenu.cpp:2967:34
    frame #16: 0x00007ffff4f2d919 libQt5Widgets.so.5`QWidget::event(this=0x00007fffffffb810, event=0x00007fffffffab40) at qwidget.cpp:8671:26
    frame #17: 0x00007ffff50a8f10 libQt5Widgets.so.5`QMenu::event(this=0x00007fffffffb810, e=0x00007fffffffab40) at qmenu.cpp:3089:26
    frame #18: 0x00007ffff4ee1813 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(this=<unavailable>, receiver=0x00007fffffffb810, e=0x00007fffffffab40) at qapplication.cpp:3640:31
    frame #19: 0x00007ffff4eea60a libQt5Widgets.so.5`QApplication::notify(this=<unavailable>, receiver=0x00007fffffffb810, e=0x00007fffffffab40) at qapplication.cpp:3084:43
    frame #20: 0x00007ffff6c8d224 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd110, receiver=0x00007fffffffb810, event=0x00007fffffffab40) at GuiApplication.cpp:83:40
    frame #21: 0x00007ffff429ef19 libQt5Core.so.5`QCoreApplication::notifyInternal2(receiver=0x00007fffffffb810, event=0x00007fffffffab40) at qcoreapplication.cpp:1064:24
    frame #22: 0x00007ffff429f1b4 libQt5Core.so.5`QCoreApplication::sendSpontaneousEvent(receiver=0x00007fffffffb810, event=0x00007fffffffab40) at qcoreapplication.cpp:1474:27
    frame #23: 0x00007ffff4ee937b libQt5Widgets.so.5`QApplicationPrivate::sendMouseEvent(receiver=0x00007fffffffb810, event=0x00007fffffffab40, alienWidget=0x0000000000000000, nativeWidget=0x00007fffffffb810, buttonDown=0x00007ffff534d470, lastMouseReceiver=0x00007ffff534d450, spontaneous=true, onlyDispatchEnterLeave=fal [... cut ...]
    frame #82: 0x00007ffff3629e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffddb8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdda8) at libc-start.c:392:3
    frame #83: 0x000055555555dc25 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35569 (Git)
Build type: Debug
Branch: copy_paste_2023
Hash: cf3aeb6618cf68af754a74ac97f094fbb8188702
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

wwmayer, 2024-03-20:
The reason of the crash is inside the function collectMovableDependencies that does a static_cast of the Sections property of AdditiveLoft001 to PropertyLinkList. However, the Sections property is of type PropertyLinkSubList and thus the static_cast causes undefined behaviour.

https://github.com/FreeCAD/FreeCAD/pull/13050 fixes the hard crash by replacing the static_cast with a dynamic_cast.

So, the remaining issue is the failing boolean operation. Boolean operation are sensitive to overlapping faces. To fix the issue you should make the Pad003 longer in both directions:
* use the mode Two lengths
* set the first length to Spreadsheet.SpikeHeight + 2
* set the second length to e.g. 1

After a recompute the Common should be as expected.