FreeCAD/FreeCAD #11990: OpenSSL version mismatch in appimage while trying to update macros from github

selkovjr, 2024-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The error is in  
`FreeCAD_weekly-builds-35570-2024-01-12-conda-Linux-x86_64-py310.AppImage`

The previous week's build also had a git error: it included its own git which did not recognize the submodule command 

This appimage uses my system git, which does not work with the (apparently) included SSL library.

When I run the command externally, it works (but then the result is wiped out during subsequent runs, so I can't install add-ons. when run by the add-on manager, the following error occurs:

```
Called with: /usr/bin/git clone --recurse-submodules https://github.com/FreeCAD/FreeCAD-macros.git /home/selkovjr/.cache/FreeCAD/Cache/AddonManager/Macros

Returned stderr:
Cloning into '/home/selkovjr/.cache/FreeCAD/Cache/AddonManager/Macros'...
OpenSSL version mismatch. Built against 30000020, you have 30200000
```

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (Unity:Unity7:ubuntu/unity)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35570 (Git) AppImage
Build type: Release
Branch: main
Hash: 3ba0c3d7958dad891e125e065a41d95faac47c6e
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-15:
I'm struggling to replicate this, @adrianinsaval as far as I can tell OpenSSL has changed recently in the world of Conda?

selkovjr, 2024-01-15:
Just to confirm, I built the FreeCAD binary from the current head,
fa0a25a777, and it works without problems. Something is wrong inside
appimage.

adrianinsaval, 2024-01-26:
I will get my hands on mac device during fosdem and will attempt to reproduce this there

luzpaz, 2024-08-24:
@adrianinsaval any results ?

maxwxyz, 2024-11-01:
ping

adrianinsaval, 2024-11-01:
I was never able to reproduce this, is this still a thing?

adrianinsaval, 2024-11-01:
> The previous week's build also had a git error: it included its own git which did not recognize the submodule command

AFAIK we never shipped git in the appimage so this seems incorrect, there could be something wrong in the system. I think we could ship with git though and maybe that would actually be better

luzpaz, 2025-06-10:
@selkovjr is this still an issue ?