FreeCAD/FreeCAD #11991: [Problem] Addon Manager shows installed and available versions are the same for updates

DavidGriffith, 2024-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I started the Addon Manager to see what workbenches have updates available, I noticed that the five which have updates available, the version I have and the updated version are the same.  Like this: 
````
Installed version: 0.4.59...   Available version: 0.4.59
````

I expected to see the `Available version:` field showing some higher value.

[image]

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (MATE/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35580 (Git)
Build type: Release
Branch: main
Hash: 2f710755bc5a1594e45a953b4d1082edd131fb65
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * sheetmetal 0.3.0
  * freecad.gears 1.0.0
  * Assembly4 0.50.2
  * A2plus 0.4.60n
  * fasteners 0.4.66
  * Manipulator 1.5.0
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-15:
Based on the assumption that the screenshot was taken very shortly before logging the 'problem' then I would say you haven't clicked the button in the bottom left corner of that window - _Refresh local cache_ for a good while as the version number quoted for Sheetmetal in particular is way behind the current one.

DavidGriffith, 2024-01-15:
Oops!  I hadn't noticed this behavior before.

closed 2024-01-15 by DavidGriffith