FreeCAD/FreeCAD #11995: [Problem] Creating an additive helix using the face from a previous helix has unexpected results

Anthony-Gaudino, 2024-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I create an additive helix in PartDesign and then I select the face of the end of this helix and use it to create a new helix it seems that FreeCAD creates the helix with about half coil behind where it should be.

I expect the second helix to begin exactly from the selected face.

Here's an example that shows the issue: [spring_problem_2.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13941387/spring_problem_2.FCStd.zip)

[image]

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.22
  * ThreadProfile 1.89.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.12.28
  * Silk 0.1.5
  * Defeaturing 1.2.1
  * A2plus 0.4.61
  * Manipulator 1.5.6
  * fasteners 0.5.0
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.15
  * toSketch 1.0.1
  * Curves 0.6.22
  * Assembly4 0.50.6
  * Assembly3 0.12.0
```

### Subproject(s) affected?

PartDesign

### Anything else?

Maybe this issue only happens if the face selected belongs to a helix that has a cone angle value or having a cone angle simply makes the issue more apparent.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Anthony-Gaudino, 2024-07-12:
Updated Full version info as it still happens on 0.22.0dev.

maxwxyz, 2024-08-27:
@FlachyJoe FYI

ElementW, 2026-09-15:
Could reproduce on (all versions given are x86_64 AppImages) 0.20.0, 0.21.2, but from 1.0.0 onwards, i.e. all stable and the few random weeklies I've tested, the problem mentioned stopped happening: you can create a helix from the tip of another one just fine.

Previous versions failed because OCC created a degenerate face covering the tip, but 1.0.0 upgraded to OCC 7.6.3 and no longer does this.

Another issue lingers here though: the 2nd helix starts from `AdditiveHelix.Face6` which doesn't exist any more making recompute fail, but the helix code misidentifies this as a null shape instead of reporting `AdditiveHelix001: Sub shape not found: spring_problem_2#AdditiveHelix.Face6` like a Pad does for instance.

@maxwxyz Strictly speaking this bug was fixed a while ago but unearths a UX issue still in `main`. I found no existing issue for the latter, should I open it and we close this one?

maxwxyz, 2026-09-16:
OK

closed 2026-09-16 by maxwxyz