FreeCAD/FreeCAD #12006: [Problem] If I set the language settings in Korean, I will no longer be able to access preferences.

sumin0514, 2024-01-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm using freecad in mac os sonoma.
The version is 0.21.2-33771.
set my freecad's language to Korean, I will not be able to access preferences on the menu or with shortcuts.

### Full version info

```shell
I'm using freecad in mac os sonoma.
The version is 0.21.2-33771.
set my freecad's language to Korean, I will not be able to access preferences on the menu or with shortcuts.
change the settings to English using the python console, it will be accessible again.
As far as I know, similar problems have been reported in other languages
Is it possible to fix this problem?

This is my first issue report, so it may not fit the form.
thanks.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-16:
It's not reproducible on Ubuntu. If I encounter it, I will just remove the settings completely. =)

By the way, could you completely remove the settings, set Korean and see what happens? I know how to do this on Windows or Linux, but not on MacOS.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35569 (Git)
Build type: Debug
Branch: copy_paste_2023
Hash: cf3aeb6618cf68af754a74ac97f094fbb8188702
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

Roy-043, 2024-01-16:
Apparently on macOS the Preferences... menu option can appear in the FreeCAD menu instead of the Edit menu. Have you checked both menus?

chennes, 2024-01-20:
Didn't we have this same issue with another language on Mac OS? I recall looking into it, but not finding any obvious culprits.

adrianinsaval, 2024-01-26:
I think there where two menu commands that had the same translation and that caused some problem, but I could be wrong. I think it was polish or russian

fefefico69, 2024-05-04:
Ran into the same problem in italian version.
Configuration menu gives these 4 choices which don't do anything when clicked.
I've **tried switching workbench** and then tried again to open preferences and the whole program freezes and needs to be force-quitted.

[image]

maxwxyz, 2024-11-09:
The same issue was reported for Russian language on macOS: https://github.com/FreeCAD/FreeCAD/issues/17714

gennua, 2025-02-26:
Hello Guys,

have the same issue with preferences in Ukrainian language! Mac OS 15.3, FreeCAD 1.0.0
described it here https://forum.freecad.org/viewtopic.php?t=95120

chennes, 2025-09-26:
If so, it will almost certainly be fixed by https://github.com/FreeCAD/FreeCAD/pull/24014

closed 2025-11-24 by luzpaz