FreeCAD/FreeCAD #12008: [Problem] Multiple Monitors - Start up hangs at white screen - Reproducible

shadowfoxish, 2024-01-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On my multi-monitor computer setup, FreeCAD sometimes hangs on start up with a non-responsive white screen and needs to be End-Tasked. This is not the OpenGL/Soft-Render bug. This is something else I also experienced in an old version of Cura (Sovol Cura 1.5.6), for my 3D printer. 

Setup: Windows 10, NVidia RTX Graphics Card _only_, 2 monitors attached to the graphics card (this is not a thunderbolt connected screen, but I don't know if that matters or not). This is not a laptop that with the dual graphics cards. That's got its own problems.

Background: Windows remembers what screen a program was last closed on in the non-maximized size view. 

Screen layout, for reference:
`[S1*][S2]` S1 is primary

How to reproduce the problem:
_Don't maximize the FreeCAD window during these steps_

1) Start up FreeCAD. I've been double-clicking a FCStd file.
[image]
2) Make sure the window is restored down, (small, not maximized). Put it on [S1].
3) Close FreeCAD by pressing the X
4) Double click the FCStd file again. FreeCAD should open normally.
5) Drag FreeCAD window to monitor [S2].
6) Close FreeCAD by pressing the X
7) Double click the FCStd file again. FreeCAD will attempt to open, but you will just see this. For me, its also trying to open partially between both screens. 
[image]
8) End-Task the hung FreeCAD
9) You can do 7-8) again, it won't change now, its broken.

How to work around it:
1) Disable screen [S2] from windows Display Settings
[image]
2) Start FreeCAD. It will open on [S1] and be fine. Close it again.
3) Turn your [S2] monitor back on.
4) Open FreeCAD again, it will continue starting up on [S1] until you move it to [S2] and forget.

I thought this might have been the openGL/hardware acceleration bug that is mentioned in the known issues page, but that's not the case. Without changing that setting, I can get FreeCAD to work or not work by following the above instructions.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.22
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-01-19:
this problem was reported before: #7070
we never figured out the cause or a fix but I was not able to reproduce.

Could you test running the weekly builds? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

if that doesn't work try removing %APPDATA%/FreeCAD and check if it works after multiple launches

shadowfoxish, 2024-01-19:
Thanks for getting back to me. I unzipped that and ran it, and ran through the steps and it's also exhibiting the behavior. I cleared out the APP DATA folder as you mentioned as well. 

[image]

Do you have any ideas for a log or anything? It might not be super relevant here, but I do have visual studio on this machine as well, I could possibly poke some stuff deeper with a bit of guidance

adrianinsaval, 2024-01-20:
You can try running `freecad --log-file C:/some/path/freecad-log.txt` it should write a log to the given path but I'm not sure it'll give much relevant info.

If you want to debug, you'll have to ask someone else for help I don't have experience making debug builds on windows, and there's a problem with the current windows libpack that is making debug builds fail.

Syres916, 2024-01-20:
@shadowfoxish you mentioned that you had a similar problem with an older version of Cura which also uses Qt5.15.x. So a quick look in the Cura GitHub Issues has a possible solution and explains why some uses can't replicate the issue because they don't have the quoted service see https://github.com/Ultimaker/Cura/issues/12197#issuecomment-1200322553

shadowfoxish, 2024-01-20:
@Syres916 I'll be darned. Turning off NahimicService fixed the issue. Seems to have worked for both FreeCAD and Cura.

Syres916, 2024-01-20:
I've asked the custodian of the Known Issues page on the forum to update it to include this. I don't know what else we can do to be honest so if you're happy are you OK to close this?

shadowfoxish, 2024-01-20:
Yeah, I think that's totally fine. This is obviously not your issue, but huge appreciation for the tip on Nahimic.

closed 2024-01-20 by shadowfoxish