FreeCAD/FreeCAD #12016: [Problem] Sketcher: Bspline tool could use OVP and widget

PaddleStroke, 2024-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue aims to discuss the specific of rework of the bspline tool to take advantage of tool widget and OVP.

There is a general issue about all the tools that needs to be updated : https://github.com/FreeCAD/FreeCAD/issues/11290

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-01-19:
Maybe Also the different view options for bsplines could be added at the bottom of the widget, when the bspline tool is active. Just with checkboxes, e.g. shiw degree, show curvature comb,...

adrianinsaval, 2024-01-19:
> Maybe also the different view options for bsplines could be added at the bottom of the widget, when the bspline tool is active. Just with checkboxes, e.g. show degree, show curvature comb,...

not sure if that makes sense given that those thigns are not shown while editing a bspline

AjinkyaDahale, 2024-01-20:
> > Maybe also the different view options for bsplines could be added at the bottom of the widget, when the bspline tool is active. Just with checkboxes, e.g. show degree, show curvature comb,...
> 
> not sure if that makes sense given that those thigns are not shown while editing a bspline

Curvature comb might be an overkill, but degree is definitely something we can consider showing. We can also change degree so knowing that would be useful.

PaddleStroke, 2024-05-16:
fiy I'm on it