FreeCAD/FreeCAD #12020: ISO GPS

sebaras, 2024-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

ISO GPS is a general Problem in TechDraw
For eg.:
The Symol (E) should be after the tolerance
There are other symbols and combines symbols missing. See examples in attachements

### Full version info

```shell
[code]
OS: Windows 10 build 19042
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Switzerland (de_CH)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * CADExchanger
  * ExplodedAssembly
  * fasteners 0.4.60
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-01-17:
Is this a duplicate of #11112? Seems to be some overlap at least.  

There are also corresponding standards in ASME Y14.5.

pierreporte, 2024-01-17:
@WandererFan Definitely a duplicate. GPS is a synonym of GD&T.

adrianinsaval, 2024-01-19:
closing as duplicate, feel free to repon if it turns out it's not actually a duplicate

closed 2024-01-19 by adrianinsaval