FreeCAD/FreeCAD #12021: [Problem] There is no solidworks navigation available in the Edit>Preference>Display>Navigation>Navigation>3D Navigation

hemangjoshi37a, 2024-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want to use solidworks style navigation but there is no option for this in the Preference Dialogue Box.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.3.16
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-17:
Here is an example of a new navigation style submitted in the past https://github.com/FreeCAD/FreeCAD/commit/c397aee9ed3efcb2e33fa719313c98cc4867cf32
FreeCAD users wouldn't necessarily even know what the Solidworks nav style is.

hemangjoshi37a, 2024-01-17:
> Here is an example of a new navigation style submitted in the past https://github.com/FreeCAD/FreeCAD/commit/c397aee9ed3efcb2e33fa719313c98cc4867cf32
> FreeCAD users wouldn't necessarily even know what the Solidworks nav style is.

It is not freecad users to know but it is for the users that are coming from solidworks background. It makes it easy to transition. Also the suggestions should be welcomed I think in an opensource project.

Syres916, 2024-01-17:
> It is not freecad users to know but it is for the users that are coming from solidworks background.

I'm well aware who it's _for_ but it's the community who submit the Pull Requests, how is anyone form the community supposed to create a Nav Style for a product I certainly have never used and for that matter never want to. I welcome your Pull Request preferably or at the _**very least**_ a full run down of each mouse function and what it's supposed to transpose to such as Pan, Rotate etc.

howie-j, 2024-01-17:
@hemangjoshi37a I find the `blender` navigation style to be the most similar to solidworks. Give it a try.

Rexbas, 2024-01-18:
> or at the very least a full run down of each mouse function and what it's supposed to transpose to such as Pan, Rotate etc.

Looks like that is described here: https://help.solidworks.com/2024/English/Visualize/c_using_SW_camera_navigation.htm?verRedirect=1

I never used Solidworks so no idea if it is a complete list.

hemangjoshi37a, 2024-01-19:
@Rexbas your commented link is not what we intend but it is a good start to understand other 3d design software's navigation system. also if someone coming from another software then they should feel at least somewhat familiar with UI so that they can easily migrate. But anyways it is a good start and thanks for your feedback .

Syres916, 2024-01-19:
There is no way anyone should create a Solidworks nav style unless it's first been documented by at least two current SW users. If we add a nav style that's just going to generate a load of complaints on the forum it's no good for anyone. As the OP doesn't want to actually _contribute_ then this should be marked as low priority and I'll contact a helpful SW user locally or on the forum. Only when the document has been agreed then coding should begin.

FEA-eng, 2024-01-20:
> Looks like that is described here: https://help.solidworks.com/2024/English/Visualize/c_using_SW_camera_navigation.htm?verRedirect=1

This link points to SolidWorks Visualize navigation. Here's what it looks like for SolidWorks itself:
https://help.solidworks.com/2024/English/SolidWorks/sldworks/t_pan_fundamentals.htm
https://help.solidworks.com/2024/English/SolidWorks/sldworks/t_roll_view.htm
https://help.solidworks.com/2024/English/SolidWorks/sldworks/t_rotate_view.htm#qnp1450446507501
https://help.solidworks.com/2024/English/SolidWorks/sldworks/t_zoom_in_out.htm

@Syres916 As a SolidWorks user, I can help if needed. Personally, I would like to see a few additional navigation styles including SolidWorks one (https://forum.freecad.org/viewtopic.php?t=66882) but I know that it requires quite some effort. Too bad it's not just a matter of specifying the buttons used for 4 basic operations (select, pan, zoom and rotate).

chennes, 2024-01-20:
The Issue here is perfectly reasonable -- I would love ways to make it easier for users of other commercial packages to give FreeCAD a try, and making the navigation style they are used to easily available would be a good step in that direction. Presumably @hemangjoshi37a is willing to help, and we have @FEA-eng as well -- what's needed is for someone to take an existing Nav style, copy it, rename it, and adjust the contents as needed. It is under 400 lines of code, most of which are completely rote. I suspect even a non-developer can more or less figure it out, and I'd be happy to review a PR that closes this Issue.

FlachyJoe, 2024-01-21:
Side not: we certainly shouldn't give to any FreeCAD feature a commercial trade name.

xtemp09, 2024-01-21:
> Side note: we certainly shouldn't give to any FreeCAD feature a commercial trade name.

Liquidworks? Fluidworks?

FEA-eng, 2024-01-21:
> Side note: we certainly shouldn't give to any FreeCAD feature a commercial trade name.

Well, we already have the Revit navigation style. But maybe we could just call the new style SW and elaborate in the description/documentation.

hemangjoshi37a, 2024-01-22:
> Side note: we certainly shouldn't give to any FreeCAD feature a commercial trade name.

We can try `solid works` that is not exactly a trade name because there is a space between.

MisterMakerNL, 2024-02-09:
My day job is working with Solidworks, I use Blender navigation.
I do not miss anything to be honest.
I almost never use Visualize since Blender is way more powerful.
Also I don't see a lot of Solidworks users use Visualize since most engineers don't do nice renders. Maybe product designers, but those designers often have more specialized render software. Visualize kinda sucks in my opinion. 

Being more clear about it that Blender is also similar to Solidworks in navigation would be a good idea. 
CAD style is weird tho, it suggest it is for CAD users. But in fact it is the og Freecad style.
I wonder how many people use this style.

My suggestion:

- Rename Blender to Blender/Solidworks
- Rename CAD to FreeCAD.

maxwxyz, 2024-03-09:
@MisterMakerNL CAD us also used in CATIA so most which are familiar with CATIA or use CATIA on the day job will use this navigation, I myself. Maybe rename CAD to CATIA as I am not sure what other CAD software uses this style.

maxwxyz, 2024-03-09:
to add from the FreeCAD Day 2024 Complaint Session:
**We have multiple navigation presets which don’t match other CAD tools.**
For example, no “ONSHAPE”, Solidworks, Fusion, Inventor etc

hemangjoshi37a, 2024-03-13:
@MisterMakerNL I don't know if you have customized your control settings for blender or what because solidworks control and blender control are so much different in many ways.

closed 2025-02-24 by chennes