FreeCAD/FreeCAD #12025: [Problem] Unable to install freecad

jelmd, 2024-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Unfortunately it is not possible to install free-cad on Ubuntu 22.04 (jammy) using the [PPA Launchpad](http://ppa.launchpad.net/freecad-maintainers/freecad-daily/ubuntu) because of `freecad-daily-python3 : Depends: python3-pyside2.qtsg but it is not installable`. The actually reason seems to be, that this package has not been built so far on the PPA. So is it possible to submit the required jobs to let it [and possibly others] be build.

### Full version info

```shell
not installed yet
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

silentjet, 2024-01-18:
The same story since several days at least.
It seems it is already fixed, just packages not yet rolled out:
https://salsa.debian.org/adrianinsaval/freecad/-/commit/655e9c801547e4111c9b64077be7ee1f02946ebd

tximikel, 2024-01-22:
> The same story since several days at least. It seems it is already fixed, just packages not yet rolled out: https://salsa.debian.org/adrianinsaval/freecad/-/commit/655e9c801547e4111c9b64077be7ee1f02946ebd

I've the same problem, do you know when will be released fixed packages? or how i can fix it? thanks!

silentjet, 2024-01-22:
No, I don't.
But you can install bit "older" FreeCAD from snaps or recompile freecad package on your own (on my laptop it took maybe 15-20 minutes). It is pretty much straight formward procedure

jelmd, 2024-01-25:
> No, I don't.

Why not?

> But you can install bit "older" FreeCAD from snaps

The snaps stuff is the biggest bullshit in the world. It copies several hundred MB of each package (e.g. firefox ~400 MB) into __each__ users home and often fails miserably because it is totally dumb, has no clue of NFS. So really nice for 1000+ students, which all have their homes on NFS and nice for the DC because of the huge redundant data aka junk this poorly designed software produces. No thanx, snapd is banned on our systems (and therefore all packages requiring it). 

> or recompile freecad package on your own (on my laptop it took maybe 15-20 minutes). It is pretty much straight forward procedure

And what about all its dependencies like *pyside* stuff? Perhaps we should consider to drop freecad, too.

silentjet, 2024-01-25:
> > No, I don't.
> 
> Why not?

Because I'm not a package maintainer thus I do not know when there would be new package version

> The snaps stuff is the biggest bullshit in the world. It copies several hundred MB of each package (e.g. firefox ~400 MB) into **each** users home and often fails miserably because it is totally [...]

My Chrome cache is 18G big. Oh well, I don't care if snap brings additional 400Megs, really. As for the rest, I'm not a fan of snaps, but if there is no other option so let's be it.

> > or recompile freecad package on your own (on my laptop it took maybe 15-20 minutes). It is pretty much straight forward procedure
> 
> And what about all its dependencies like _pyside_ stuff? Perhaps we should consider to drop freecad, too.
Not sure which particular problem you are reffering to. Recompilation solved all "my" issues... The only thing I did extra, probably, lowering down version of libocct* (and related packages) to the version from Jammy.

lcpt, 2024-01-25:
Hi.

I have the same problem too. It is possible to install those dependencies "manually"?. I mean, are the required packages stored somewhere in `http://ppa.launchpad.net/freecad-maintainers/`?

adrianinsaval, 2024-01-26:
ppa has been updated and is now working

closed 2024-01-26 by adrianinsaval

jelmd, 2024-01-26:
> ppa has been updated and is now working

@adrianinsaval - Thanx a lot! Any intention to update https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-stable as well?

adrianinsaval, 2024-01-26:
Yes it'll be updated soon

eshazen, 2024-04-08:
It's April 8, and I still can't install on Ubuntu 22.04.

hazen@botlab-roblyer-hazen 3D (main) $ sudo apt install freecad
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libocct-data-exchange-7.6 : Breaks: libocct-data-exchange-7.5 but 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 is to be installed
 libocct-foundation-7.6 : Breaks: libocct-foundation-7.5 but 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 is to be installed
 libocct-modeling-algorithms-7.6 : Breaks: libocct-modeling-algorithms-7.5 but 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 is to be installed
 libocct-modeling-data-7.6 : Breaks: libocct-modeling-data-7.5 but 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 is to be installed
 libocct-ocaf-7.6 : Breaks: libocct-ocaf-7.5 but 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 is to be installed
 libocct-visualization-7.6 : Breaks: libocct-visualization-7.5 but 1:7.5.2+dfsg1-0~202107020155~ubuntu22.04.1 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
hazen@botlab-roblyer-hazen 3D (main) $

jelmd, 2024-04-08:
@eshazen did you try to remove the 7.5.2 packages first?

Install worked for me:
```
ii  libocct-data-exchange-7.6:amd64       1:7.6.3+dfsg1-8~ubuntu22.04.1               amd64        Open CASCADE Technology module for CAD data format interoperability
ii  libocct-draw-7.6:amd64                1:7.6.3+dfsg1-8~ubuntu22.04.1               amd64        Open CASCADE Technology command interpreter & graphical test library
ii  libocct-modeling-algorithms-7.6:amd64 1:7.6.3+dfsg1-8~ubuntu22.04.1               amd64        Open CASCADE Technology geometrical & topological algorithms module
ii  libocct-modeling-data-7.6:amd64       1:7.6.3+dfsg1-8~ubuntu22.04.1               amd64        Open CASCADE Technology 2D/3D geometric primitives data structures
ii  libocct-ocaf-7.6:amd64                1:7.6.3+dfsg1-8~ubuntu22.04.1               amd64        Open CASCADE Technology application-specific data and functionality
ii  libocct-visualization-7.6:amd64       1:7.6.3+dfsg1-8~ubuntu22.04.1               amd64        Open CASCADE Technology graphical data visualization module
ii  freecad                               2:0.21.2+dfsg1~202401262230~ubuntu22.04.1   all          Extensible Open Source CAx program
ii  freecad-common                        2:0.21.2+dfsg1~202401262230~ubuntu22.04.1   all          Extensible Open Source CAx program - common files
ii  freecad-python3                       2:0.21.2+dfsg1~202401262230~ubuntu22.04.1   amd64        Extensible Open Source CAx program - Python 3 binaries
```

`/etc/apt/sources.list.d/freecad.list`:
```
deb http://ppa.launchpad.net/freecad-maintainers/freecad-stable/ubuntu jammy main
```