FreeCAD/FreeCAD #12027: [Problem] Path: Error no "ToolController" with combined dressups

ModuloFS, 2024-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,

i have mentioned a problem with a LeadInOut dressup after a Tag dressup in this post some time ago: [viewtopic.php?t=80507](https://forum.freecad.org/viewtopic.php?t=80507) (my last post in this thread)
This problem is still present and you can find an example in the attached file. If i try do add the LeadInOut dressup on top of the Tag dressup made on Profile i get the following error:

`Running the Python command 'Path_DressupLeadInOut' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 724, in Activated
    FreeCADGui.doCommand("dbo.setup(obj)")
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.21\Mod\Path\Path\Dressup\Gui\LeadInOut.py", line 147, in setup
    obj.Length = obj.Base.ToolController.Tool.Diameter * 0.75

'FeaturePython' object has no attribute 'ToolController'`

The LeadInOut dressup is created standalone above the Tag dressup in the tree. It is not editable, if you made a double click on it you end in the transformation window...

The Forum post to this problem is under: https://forum.freecad.org/viewtopic.php?t=84339

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.61
  * btl 0.9.9
  * fasteners 0.4.71
  * FeedsAndSpeeds 0.5.0
  * Lithophane
  * Manipulator 1.5.0
  * OpenDark 2023.11.27
  * QuickMeasure 2022.10.28
  * sheetmetal 0.3.12
```

### Subproject(s) affected?

None

### Anything else?

[image]
[Multible_Dressup_Error.zip](https://github.com/FreeCAD/FreeCAD/files/13969425/Multible_Dressup_Error.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-06-02:
@ModuloFS is this still relevant ?

ModuloFS, 2025-06-03:
I will try it again in the next days with a current weekly version.

tarman3, 2025-08-03:
Can not reproduce

davidgilkaufman, 2025-08-22:
@sliptonic I'm pretty sure this bug no longer exists, let's close the issue