FreeCAD/FreeCAD #12029: Path : Value of Split output>Order Output By not kept when opening the job (Freecad in French)

rickenbroc, 2024-01-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad 0.21 and 0.22.dev set in FRENCH
I've defined a job and set the split output option to "Outil" (Tool in French)
When looking at the properties panel, the "Order Output By" parameter is set to "Tool" (displayed in English)
When opening the job, the split output option is reset to "Fixation" instead of keeping it to "Outil" (seems like FC can't find the French term and put it back to its default value)
If I save the job, then the "Order Output By" is set to "Fixture"

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git)
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * btl 0.9.9
  * CurvedShapes 1.0.4
  * Curves 0.6.12
  * fasteners 0.4.56
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * ImportNURBS 1.1.0
  * Marz 0.0.27-alpha
  * QuickMeasure 2022.10.28
  * workfeature
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

rickenbroc, 2024-01-19:
Thx Syres916,
Maybe this bug opening will stand out enough for capting the attention of @sliptonic.

sliptonic, 2024-01-19:
https://github.com/FreeCAD/FreeCAD/pull/12041

commit 30b907ec (2024-01-19): Merge pull request #12041 from Ondsel-Development/fixorderbywidget

fixes #12029
commit ccaa636c (2024-01-19): fixes #12029
closed 2024-01-19 by sliptonic