FreeCAD/FreeCAD #12030: [Problem] Mouse navigation: some styles have rotation issue in sketcher

LemonBoy, 2024-01-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some mouse navigation styles employing the left and/or the right mouse button for panning/zooming/rotation seem to trigger the rubber-band selection or the context menu when the mouse buttons are released.

The same problem was reported in #10624 for the TinkerCAD style and fixed in #11841 but the problem does apply to the other styles aswell.

The fix should be applied to the other classes and, since the code looks quite similar across them, is probably better off in a separate method.

cc @reba

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Rexbas, 2024-01-18:
#11841 and #11855 fixed some of these issues but were only merged in last two weeks and is not included in 0.21.x. I will take another look. If you want you can try the weekly build (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) which includes these fixes.

Besides TinkerCAD and OpenSCAD style, are there any navigation styles in particular that are not working as expected or is this a problem in all navigation styles? And which key/mouse combinations trigger the rubber-band selection or context menu for you? Maybe I missed some when I tested the fixes.

LemonBoy, 2024-01-18:
> Besides TinkerCAD and OpenSCAD style, are there any navigation styles in particular that are not working as expected or is this a problem in all navigation styles? And which key/mouse combinations [...]

The problem shows up for pretty much every other style:
- The two-button pan in "Blender" style triggers both the rubber-band selection and the context menu depending on which key is pressed first.
- The pan/zoom mode using Ctrl/Ctrl+Shift in "CAD" style reliably opens the context menu
- Other styles are affected, seemingly where the ctrl/alt is used combined with a mouse movement and/or two mouse buttons are involved. If required I'll go trough all the styles and write an extensive list.

Rexbas, 2024-01-18:
Thanks, I can reproduce it. I will also do some testing myself and although it is not required, an extensive list would certainly help to make sure we are not missing any cases.

PR #12104 (2024-02-12): Gui: Mouse navigation fixes
Fixes #12030.

This PR consists of 3 types of fixes. The first commit makes sure the right click context menu won't be shown after dragging, panning or zooming. The second commit prevents starting the sketcher rubber-band selection while another navigation button is pressed. The third commit prohibits switching to another mode while the rubber-band selection is active.
closed 2024-02-12 by chennes