FreeCAD/FreeCAD #12036: [Feature request] Sketcher: more visual differences for dimensions

pierreporte, 2024-01-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are 4 kinds of dimensions in sketcher:

1. normal dimensions,
2. reference dimensions,
3. dimensions driven by an expression,
4. deactivated dimensions.

Currently, only color is used to differentiate them. It has been explained in #10100 that the number of colors should be reduced so it is necessary to find a way for entities to be visually different without color.

Engineering drawing define a few things about this and the industry already solved the problem:

* Dimensions that do not define the geometry and are present for information only have parentheses around them. This is exactly what references dimensions are in FreeCAD sketches. Seasoned users will recognize instantly what the dimension is. This solution will allow for reference dimensions to have the same color as normal dimension.
* Dimensions that are driven could have an icon or small text near the value to indicate that. It is a solution widely used in the industry. Solidworks has a Σ icon, Catia has also a small icon like *f(x)*, Fusion 360 has a `fx:` prefix. Adding something (icon or text) will allow these dimensions to have the same color as normal and reference dimensions.
* A new kind of dimension has been requested for arc length in #9924. Instead of having yet another color, they should define their equivalent in engineering drawings, that is having an arc above or before the value. The symbol already exists in osifont which is bundled with FreeCAD.
*  Deactivated dimensions are muted and I believe they can stay so, using the same color but darker or lighter depending on the theme.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

[image]

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-01-19:
> * Dimensions that do not define the geometry and are present for information only have parentheses around them. This is exactly what references dimensions are in FreeCAD sketches. Seasoned users [...]
 
+1

> * Dimensions that are driven could have an icon or small text near the value to indicate that. It is a solution widely used in the industry. Solidworks has a Σ icon, Catia has also a small icon like [...]

+1
we use the _f(x)_ icon for expressions on FreeCAD so this is probably what should be used

> * A new kind of dimension has been requested for arc length 

I don't think anyone would think of using different colors to differentiate these. They are not a new category of constraints, just new constraints. On the matter of differentiating from arc angle showing the unit of measure is probably enough.

> * Deactivated dimensions are muted and I believe they can stay so, using the same color but darker or lighter depending on the theme.

On the contrary, deactivated constraints would be more useful if they would actually still display in 3d view, they could have ~strike through~ text though

pierreporte, 2024-01-19:
> I don't think anyone would think of using different colors to differentiate these. They are not a new category of constraints, just new constraints. On the matter of differentiating from arc angle [...]

The symbol makes the dimension clearer in situations when the arc has a large radius and a short chord. It won’t be the most used constraint so it wouldn’t overload the view.

> > Deactivated dimensions are muted and I believe they can stay so, using the same color but darker or lighter depending on the theme.
>
> On the contrary, deactivated constraints would be more useful if they would actually still display in 3d view, they could have strike through text though

Strike through, visible text is a good idea but they should still be muted (maybe not that muted but still).

adrianinsaval, 2024-01-19:
if we muted you mean a gray color I agree, I thought you meant they should disappear as they currently do

pierreporte, 2024-01-30:
@PaddleStroke what do you think of this? It goes with the dashed lines #11996 and points styles #11919.

PaddleStroke, 2024-02-01:
Using fx: or something like this for expression constraint is a +1 for me.

Using same color for normal and driven and gray for deactivated : I don't like this so much. Imo driven constraint and deactivated should merge into a single concept. A reference is basically a unactivated constraint. 
Then we would have red constraints and gray unactivated constraints/references. It would semantically make a lot of sense because then people would understand at once that gray constraints are not active. As grayed out is the color for deactivated things.

But that was already discussed at length on the forum and it got very polemic so I dropped this subject.

pierreporte, 2024-02-01:
I’m not against merging reference and deactivated dimensions but in this case they should both have parentheses. But dice there is already 3 kind of dimensions, does that mean than one of them will be deprecated or this would be done by style only?

pierreporte, 2024-02-14:
@PaddleStroke are you okay with parentheses for reference dimensions?

DrD4ffy, 2024-02-15:
What about a mockup how it could look like:

1. On the right a normal dimension named 'height'
2. On the top an expression driven dimension (2* height)
3. On the left a reference dimension
4. On the bottom a deactivated dimesion

[image]

Some explanaitions:
- Minimalistic color usage: Everything that in fact describes the actual dimension of an element (1, 2 and 3) is in red, the deactivated dimension is gray and striked out.
- Each dimension can keep its own color property in Preferences so users who like colorful sketches can still use colors, but the default should be one (two with deactivated) color(s)
- Previously I proposed 'f(x) = 40 mm' for calculated dimensions, but f(x) could be a name. '40 mm (fx)' is unambiguous.
- The value of the deactivated dimension (4) is visible, so you can select/grab it. Right now this is nearly impossible.
- Don't merge the concept of reference and deactivated dimension, they are NOT the same. Deactivated dimensions can have a completely differe value (see my sketch) that is just not applied right now. Personnaly I don't use deactivated dimensions, but the point is that right now FreeCAD has the concept of deactivated dimensions, and people might use it to temporarily get some DoF and reactivate it later without loosing the information about what length it should be. Discussing this here is sort of scope creep preventing the 'improve visual differences for dimensions' from being finished. If you don't need it, just don't use it.

pierreporte, 2024-02-15:
Your proposition looks perfect.

Dimensions.png (view on web)

pierreporte, 2024-02-24:
@FlachyJoe I just saw that the arc length constrain is on [your TODO list](https://github.com/users/FlachyJoe/projects/1/views/1?pane=issue&itemId=38885600). You may be interested in this special dimension style.

FlachyJoe, 2024-02-24:
@pierreporte I don't see any reason to add a symbol to arc length label because the base line will be a curve and will be differentiate. 
Thank you to bump me to this todo list, I'm spreading my work too much.

pierreporte, 2024-02-24:
@FlachyJoe When using imperial measurement or when units are hidden, it can be hard to distinguish an arc length from an angle. It can also pave the way for a more general curve length where it would be cumbersome—if desirable at all— to follow the profile.

FlachyJoe, 2024-02-25:
@pierreporte it's easier than I thunk
[image]

pierreporte, 2024-02-25:
Great! It will look perfect when the spacing of the dimension line for the arc and unit is adjusted.

maxwxyz, 2024-03-23:
To add from https://github.com/FreeCAD/FreeCAD/issues/12756#issuecomment-1981648341
The render order should be improved to improve visibility and legibility.

1. geometry on top (would solve the overlay issues of dimension lines); dotted geometry (construction, external over normal one to see dashes)
3. constraints / dimensions
4. origin and axis cross

pierreporte, 2024-04-22:
@maxwxyz for the UI/UX project.

closed 2025-11-02 by kadet1090