FreeCAD/FreeCAD #12049: [Problem] no quick way to see basic measurements on selection

PaddleStroke, 2024-01-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Getting basic information about things is a basic need in design :  edge length, selected circle/arc radius, distance between 2 selected vertices, volume of a solid, area of a face ...

There is no easy way to get all of these easily in core. And so this feature has been implemented in many addons / workbench / branches. A few coming to mind is FCInfo, FCInfo toolbar, quick measure, RT branch has something like this too. etc.

My proposal is the same as the one described by the OP there : https://forum.freecad.org/viewtopic.php?style=4&t=80919

ie to add a label in the status bar with useful information for example : 
[image]

This label would show meaningful information about the selection depending on what it is : 
1 point : coordinates
2 points : distance
1 line : length
1 face : area
etc.

This has been discussed a little bit in https://github.com/FreeCAD/FreeCAD/pull/11344

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-01-20:
I agree that this is a problem, and I like your solution -- @obelisk79 has the Design Working Group considered this issue? Is there some advice in the Handbook that we can look to?

WandererFan, 2024-01-20:
This sounds a lot like the gsoc Unified Measurement Facility.

https://forum.freecad.org/viewtopic.php?p=733478#p733478

obelisk79, 2024-01-20:
I agree with WandererFan, there is a bit of overlap with the unified measurement facility. The DWG has discussed this kind of feature in the past. I think there is some value in displaying non-static simple information in the status bar, ie length, radius/diameter and area of flat faces. I'd probably stop there though to minimize overhead and keep behavior simple, understandable and easily predictable, leave the other type of calculations to the unified measurement functionality.

From the Design Guide:
```
Status Bar: The status bar provides contextual information about the position of the mouse cursor in relation to the objects in the main view. This section of the UI is also home to some global controls within FreeCAD. They are “Notifications”, “Navigation Style”, and “Unit Selector”. Developers shall not insert non-global elements or widgets into this area.
```

The key here being dynamic, contextual information.

DanMiel, 2024-02-16:
I would like to comment on putting the measurements on the bottom of the screen. I wrote the QuickMeasure Add-on and had the output going to the status bar. I found it awkward to move my eyes from what I was measuring to the status line and then back up to the part, I found it easier to create the existing text box and put it by what I was measuring, thereby keeping my eyes on the part and on the output.
Dan Miel

maxwxyz, 2024-02-26:
Comment from the FreeCAD Complaint Session: You can't measure the diameter of a hole if, or if you can with quick measure why is quick measure not the default measuring tool.

maxwxyz, 2024-03-02:
@PaddleStroke FYI: https://youtu.be/qPoFrgk8S2U?si=-TK0hqlpr5tTgMXm&t=1084

closed 2024-05-06 by yorikvanhavre