FreeCAD/FreeCAD #12058: Draft_ShapeString Fails On Windows With Certain Characters (UTF-8 / Unicode Related)

EmperorArthur, 2024-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

# User Issue
Draft_ShapeString throws on Windows if given a multi-line string, like "Some\nText".  Error is "EOL while scanning string literal".

While there was a [thread about a similar issue from 2020](https://forum.freecad.org/viewtopic.php?t=43842)), I found [this thread]( from 2020, this appears to be a different issue.

This appears to be caused by calling `exec` on user inputs without proper sanitation.

# How To Reproduce

On Windows, paste a multi-line string into the "String" Property of an object created by `Draft_ShapeString`.

# Second Issue

I will be reporting a security vulnerability shortly based on some additional data found while researching these Tracebacks.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.14
  * fasteners 0.4.70
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * kicadStepUpMod 10.22.8
  * MeshRemodel 1.9.22
  * parts_library
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

# Analysis

As previously noted, this appears to be an issue with passing user inputs to `exec` / Gui.doCommand.

Interestingly, while the underlying error is the same, Using the "Property" dialog produces a different Traceback, and points to multiple methods of attempting to sanitize user data.

# Traceback (Model->Property->Data)
```
20:51:55  Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'SyntaxError'>: ('EOL while scanning string literal', ('<string>', 1, 83, 'FreeCAD.getDocument(\'Plugs_And_Caps\').getObject(\'ShapeString011\').String = "Victor\n'))
20:51:55  Stack Trace: Traceback (most recent call last):
  File "<string>", line 1, in <module>
```

# Suggestion

Trying to sanitize user input to `exec` is a loosing battle.  As I noted above, it can also lead to more issues.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-25:
You mention a 2nd issue. Have you created another GitHub issue for it?

Roy-043, 2024-12-04:
New development: #16928.

Roy-043, 2025-07-01:
Another development: #22217.

luzpaz, 2026-05-24:
What does a fix for this look like ? Any proposals ?

Roy-043, 2026-10-10:
Both issues have been fixed.

closed 2026-10-10 by Roy-043