FreeCAD/FreeCAD #12060: CanAbortRecompute is True can result in a crash

Roy-043, 2024-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If CanAbortRecompute ("Allow aborting recomputation") is True, creating an Arch_Window results in an Access violation on Windows and a crash on Linux. The problem occurs in V0.22 and V0.21.2, but not in V0.20.2.

CanAbortRecompute is also one of the preferences where the default in the code does not match the default in the preferences ui file (see #10176).

Steps to reproduce:
1. Open FreeCAD.
2. Make sure the "Allow aborting recomputation" checkbox is checked.
3. Switch to the Arch Workbench.
4. Click Arch_Window.
5. Pick a point in the 3D view.
6. Result: Access violation or crash.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git)
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35639 (Git)
Build type: Release
Branch: main
Hash: f315314569aee5334fd0e69a49aab33684a802ba
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-26:
I cannot reproduce on

<details><summary>KUbuntu 22.04.3 LTS dev</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35733 (Git)
Build type: Debug
Branch: main
Hash: f75db0dbb2f48f3b486afa80afe9d75ab19970c9
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```
</details>

<details><summary>Windows 10 dev</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
</details>

<details><summary>Windows 10 0.21.1</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
</details>

<details><summary>Windows 10 0.21.2</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
</details>

Roy-043, could you provide more detailed steps to reproduce?

Roy-043, 2024-01-27:
Thanks for testing. After checking again I can't reproduce the issue consistently. I'll close this.

closed 2024-01-27 by Roy-043