FreeCAD/FreeCAD #12065: Fully constrained sketch that is interpreted in multiple ways

mjpscogman, 2024-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Background: I'm creating a switch panel for a model railroad. the switches have a washer with a tab on them. This tab is used to orient the switch in a specific direction and not allow it to move. The tabs (rectangles in the diagram) are randomly moving from their drawn position (12:00, 6:00, or 9:00) to an undesired position. This has happened at least 3 dozen times. This issue occurs in FreeCAD 0.20 and also exists in 0.21. I typically redraw them which is time consuming. 

In Sketcher draw a circle with a diameter of 0.5" centered at (0,0). Add a rectangle above the circle then add dimensions to it. On the rectangle add a point on the line closest to the circle near the middle. Then, **Constrain Symetrical** with the end points of the line. Attach that center point near the top of the circle (12:00) with **Constrain point onto Object**.  Finally select the line segment and the circle and **Constrain Tangent**.  At this point you should have a fully constrained sketch with a rectangle on TOP of a circle with the rectangle centered and attached at point (0.0, 0.5"). 

See attached "As Drawn.jpg" for a picture of what I described above

From time to time FreeCAD will move (after doing other edits saving and reopening) the box attaching it to the BOTTOM of the circle attaching it to point (0.0, -0.5")  (6:00) and occasionally to the 9:00 and 3:00 positions

The reality of the situation is more complicated. The circles are external geometry brought into the sketch through a **sub-object(s) shape binder**. The effect is still the same.  Boxes drawn at 12:00, 9:00, or 6:00 positions randomly move to other undesired positions. it is not the same boxes every time it;s not the same undesired position.

I've enclosed "Whirligig Railroad Panel.FCStd" which actually exhibits the issue. 

I've also included "Panel Squares that Moved.jpg". I've circled the switch holes and color coded them to indicate the issue as follows: 
Red - box drawn at 6:00 and moved to 12:00 or 3:00
Green - box drawn at 12:00 and moved to 6:00 or 3:00
Blue - Box drawn at the 9:00 position and moved to 6:00

All un-colored switches remain where they were drawn.

Thank you. 
[Bug Report Files.zip](https://github.com/FreeCAD/FreeCAD/files/14002090/Bug.Report.Files.zip)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.4
  * boltsfc 2022.11.5
  * kicadStepUpMod 10.21.7
  * ThreadProfile 1.89.0
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mjpscogman, 2024-01-21:
Sketch where the tabs are is called: Switch Washer Positioning Tabs

Roy-043, 2024-01-23:
This is known as 'flipping'. There is a related feature request: #10708.

I  have not fully analyzed your workflow, but since it involves creating a similar sketch and extrusion multiple times, I would suggest you look into the use of [Links](https://wiki.freecad.org/Std_LinkMake). Using them [attached](https://wiki.freecad.org/Part_EditAttachment) to edges in the master sketch can work here I think, and may be more efficient.

Using [Sketcher_ConstrainAngle]( https://wiki.freecad.org/Sketcher_ConstrainAngle) can help to prevent flipping. Please create a topic in the [FreeCAD forum](https://forum.freecad.org/) if you need more tips.

I have to close this as a duplicate.

closed 2024-01-23 by Roy-043