FreeCAD/FreeCAD #12072: [Problem] PD Subtractive_Cylinder change Body placement

FlachyJoe, 2024-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When applying a AttachementOffset to a Subtractive_Cylinder the Body placement can change an not only the cylinder one:
[image]
[image]

The problem occurs when two cylinder are in the same place.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35649 (Git)
Build type: Release
Branch: main
Hash: 35e09e233c4fadb6194c82164ece25d5043d23fe
Python 3.11.7, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```
```

### Subproject(s) affected?

PartDesign

### Anything else?

remove .zip [Cylinder_Bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14011121/Cylinder_Bug.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-23:
Can you give step-by-step instructions on how to reproduce please? Thanks.

FlachyJoe, 2024-01-23:
@Roy-043 Thank for the interest

1. Open  [Cylinder_Bug.FCStd (remove .zip)](https://github.com/FreeCAD/FreeCAD/files/14011121/Cylinder_Bug.FCStd.zip)
2. Note the main tube is extruded along Z axis
3. Set `Cylinder001.AttachementOffset.Rotation.Angle` same as one of any other Cylinders.
4. See the main tube rotated randomly

The problem occurs only if one cylinder doesn't remove any material, if its diameter is larger then the operation success.

Only the first two Subtractive_Cylinder are needed.
The problem occurs even if another feature take place between them in the tree.

Syres916, 2024-01-23:
I did test after I'd worked out the overlapped cylinders last night and went back to version 0.19.3 where the bug was still present. In order to test it on version 0.18.4 I'll need to recreate the model as there's so many bits that aren't backwards compatible. My hunch is this has existed since the 0.17dev cycle but yet to prove it.

PR #16963 (2024-10-07): PartDesign: Fix primitive placement
2276ffb fix the body movement when the AttachementOffset property is edited.
22131b3 fix the misplaced shape when the subtractive doesn't remove anything. Fix #12072 

A misplaced shape still occurs when a pattern is applied to a subtractive which doesn't remove material.

I think the error comes from `TopoShape::getSubTopoShape`. As I understand when the shape is not modified it returns the ancestor but the AttachementOffset is applied. @bgbsww @chennes @realthunder would you analyze this?