FreeCAD/FreeCAD #12078: [Problem] Path causes random python runtime errors

greggjaskiewicz, 2024-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to Edit Job in Path workbench, I sometimes am greeted with:

`21:46:36  Traceback (most recent call last):
21:46:36    File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/Path/Main/Gui/Job.py", line 167, in setEdit
21:46:36      self.openTaskPanel()
21:46:36    File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/Path/Main/Gui/Job.py", line 171, in openTaskPanel
21:46:36      self.taskPanel = TaskPanel(self.vobj, self.deleteObjectsOnReject())
21:46:36    File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/Path/Main/Gui/Job.py", line 616, in __init__
21:46:36      self.template = PathJobDlg.JobTemplateExport(
21:46:36    File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/Path/Main/Gui/JobDlg.py", line 301, in __init__
21:46:36      self.dialog.setParent(parent)
21:46:36  RuntimeError: Internal C++ object (PySide2.QtWidgets.QDialog) already deleted.`

And no dialog pops out. 

https://github.com/FreeCAD/FreeCAD/assets/553583/aea1a2a0-b036-4952-959c-1fc950034302

Had the same issue in FreeCAD weekly dev builds

### Full version info

```shell
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * AnimationFreeCAD 1.0.0 (Disabled)
  * A2plus 0.4.61
  * sheetmetal 0.3.13
  * freecad_metal_workbench 0.0.1
  * Curves 0.6.19
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-23:
Related to #9467 and #9468.

Roy-043, 2024-01-25:
Closing as duplicate.

closed 2024-01-25 by Roy-043