FreeCAD/FreeCAD #12083: [Problem] Selection handling of Sketcher_ConstrainParallel is inconsistent

Roy-043, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you select points and edges and then apply Sketcher_ConstrainHorizontal, the points in the selection are ignored. Sketcher_ConstrainParallel does not work in the same way, it refuses such a selection. IMO this is inconsistent. Although Sketcher_ConstrainParallel does not work on points, it is similar to Sketcher_ConstrainHorizontal and Sketcher_ConstrainVertical here.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35639 (Git)
Build type: Release
Branch: main
Hash: f315314569aee5334fd0e69a49aab33684a802ba
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-01-23:
@PaddleStroke check this one out, I agree it would be nice if constraint parallel would just ignore points in the selection

PaddleStroke, 2024-01-24:
Yes I can do this. I'll do it next time I checkout of branch

closed 2024-02-12 by chennes