FreeCAD/FreeCAD #12084: [Problem] Profile criteria for Helix and Revolution features are inconsitent

Roy-043, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Profile for Helix and Revolution features must be a face or a sketch. For a Pad you can also select a
PartDesign_ShapeBinder (or its edge) or a PartDesign_SubShapeBinder (or its edge or face). It is inconsistent that the same is not possible for Helices and Revolutions.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35639 (Git)
Build type: Release
Branch: main
Hash: f315314569aee5334fd0e69a49aab33684a802ba
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hkienle, 2024-01-23:
Forum discussion that triggered the issue: https://forum.freecad.org/viewtopic.php?t=84449

filippor, 2024-02-04:
Also related to bug
 #11934

PR #12207 (2024-02-26): PartDesign_Helix: use computed wires
Use the wires of the computed face instead of the selected ones.

Fix #12084