FreeCAD/FreeCAD #12091: [Problem]  SIGSEGV when exporting to 3mf

ogrady, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCad crashes when trying to export .3mf file.

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

This [file](https://github.com/FreeCAD/FreeCAD/files/14028094/tpsub.zip) causes the following error reproducibly when exporting any body or combination thereof:

```sh
~ > freecad                                                                                                                                           2m 4s
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e710) [0x7f7fcf05c710]
#1  0x7f7fd22f1464 in Gui::SoQtOffscreenRenderer::writeToImage(QImage&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0xd4
#2  0x7f7ee0912726 in MeshGui::ThumbnailExtension3MF::addMesh(Mesh::MeshObject const&) from /usr/lib/freecad/lib/MeshGui.so+0x226
#3  0x7f7ee0e5bb90 in Mesh::Exporter3MF::addMesh(char const*, Mesh::MeshObject const&) from /usr/lib/freecad/lib/Mesh.so+0xe0
#4  0x7f7ee0e5260a in Mesh::Exporter::addObject(App::DocumentObject*, float) from /usr/lib/freecad/lib/Mesh.so+0x3fa
#5  0x7f7ee0e5339d in Mesh::Module::exporter(Py::Tuple const&, Py::Dict const&) from /usr/lib/freecad/lib/Mesh.so+0x51d
#6  /usr/lib/freecad/lib/Mesh.so(+0x24c161) [0x7f7ee0e4c161]
#7  /usr/lib/freecad/lib/libFreeCADBase.so(method_keyword_call_handler+0x3e1) [0x7f7fd12c66a1]
#8  /usr/lib/libpython3.11.so.1.0(+0x1fb2e1) [0x7f7fd0dfb2e1]
#9  /usr/lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2bc) [0x7f7fd0dda8ec]
#10  /usr/lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6d3) [0x7f7fd0de4c23]
#11  /usr/lib/libpython3.11.so.1.0(+0x29c484) [0x7f7fd0e9c484]
#12  /usr/lib/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0x7f7fd0e9be6c]
#13  /usr/lib/libpython3.11.so.1.0(+0x2b9fc3) [0x7f7fd0eb9fc3]
#14  /usr/lib/libpython3.11.so.1.0(+0x2b63ea) [0x7f7fd0eb63ea]
#15  /usr/lib/libpython3.11.so.1.0(PyRun_StringFlags+0x6d) [0x7f7fd0eaaa3d]
#16  0x7f7fd1303d20 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x70
#17  0x7f7fd20eac99 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#18  0x7f7fd201ea77 in Gui::Application::exportTo(char const*, char const*, char const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x597
#19  0x7f7fd20ffe8c in StdCmdExport::activated(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0x4ec
#20  0x7f7fd20eb13f in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1ef
#21  0x7f7fd20eb4f7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x137
#22  /usr/lib/libQt5Core.so.5(+0x2d0e27) [0x7f7fcf8d0e27]
#23  0x7f7fd056bbb7 in QAction::triggered(bool) from /usr/lib/libQt5Widgets.so.5+0x47
#24  0x7f7fd057160b in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt5Widgets.so.5+0xbb
#25  /usr/lib/libQt5Widgets.so.5(+0x2f437b) [0x7f7fd06f437b]
#26  /usr/lib/libQt5Widgets.so.5(+0x2f449d) [0x7f7fd06f449d]
#27  0x7f7fd05af1a1 in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8d1
#28  0x7f7fd05788ff in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f
#29  0x7f7fd057ddaf in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x10ff
#30  0x7f7fd20b6419 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#31  0x7f7fcf89bef8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#32  0x7f7fd057c0ea in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt5Widgets.so.5+0x1ca
#33  /usr/lib/libQt5Widgets.so.5(+0x1cd9c4) [0x7f7fd05cd9c4]
#34  /usr/lib/libQt5Widgets.so.5(+0x1cec17) [0x7f7fd05cec17]
#35  0x7f7fd05788ff in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f
#36  0x7f7fd20b6419 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#37  0x7f7fcf89bef8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#38  0x7f7fcfd4196c in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt5Gui.so.5+0x6ec
#39  0x7f7fcfd2a6f5 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Gui.so.5+0xb5
#40  /usr/lib/libQt5XcbQpa.so.5(+0x65560) [0x7f7fc9cc8560]
#41  /usr/lib/libglib-2.0.so.0(+0x59f69) [0x7f7fcd90df69]
#42  /usr/lib/libglib-2.0.so.0(+0xb8367) [0x7f7fcd96c367]
#43  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x32) [0x7f7fcd90c162]
#44  0x7f7fcf8ead0c in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x6c
#45  0x7f7fcf89ac04 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x134
#46  0x7f7fcf89c0a3 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x93
#47  0x7f7fd20271cb in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x1c5b
#48  freecad(+0x4e13) [0x55ebb6315e13]
#49  /usr/lib/libc.so.6(+0x27cd0) [0x7f7fcf045cd0]
#50  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7f7fcf045d8a]
#51  freecad(+0x5fd5) [0x55ebb6316fd5]
``` 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-01-23:
Exports crashlessly. 3MF seems to be valid and contains the thumbnail.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35701 (Git)
Build type: Debug
Branch: main
Hash: 3541ccc40dee09a68ec15f029dd72d02b6fed3f7
Python 3.10.12, Qt 5.15.12, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

ogrady, 2024-01-24:
Hi @xtemp09,

thanks for the swift reply! I noticed you were using version 0.22.0, while I was on the latest release 0.21.2. Do you think this is an issue that will be fixed in the upcoming minor version, or is the culprit somewhere else (OS, Python version, locale, for all I know)?.

Best

xtemp09, 2024-01-24:
The only way to find out is to test the upcoming version. Could you test a weekly build to see if the problem remains? [^1]

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

ogrady, 2024-01-25:
> The only way to find out is to test the upcoming version. Could you test a weekly build to see if the problem remains? [1](#user-content-fn-1-0e01da15f4f9a68d9a7ce00753275d49)
> ## Footnotes
> 
>     1. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds [↩](#user-content-fnref-1-0e01da15f4f9a68d9a7ce00753275d49)

I tried the weekly build you suggested (FreeCAD_weekly-builds-35720-2024-01-23-conda-Linux-x86_64-py310.AppImage) and was able to export the model in question without issues. So I guess it's a problem with the currently released version that is already fixed in 0.22.x(?).
Thank you for looking into the matter and suggesting the weekly build. Should we close this issue or would you prefer to leave it open until 0.22.x is released?

Syres916, 2024-01-25:
As it works error free using a self compiled FreeCAD 0.22dev rather than the AppImage just to prove that Python 3.11.x wasn't having any untoward effects (I believe this export is actually coded in C++ anyway):

```
OS: Fedora Linux 38 (Thirty Eight) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35564 (Git)
Build type: Release
Branch: main
Hash: c6a020369041a229766a75d703b6c0d96951b9ab
Python 3.11.7, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * freecad.gears 1.2.0
  * fasteners 0.5.0
  * sheetmetal 0.3.15
```
I would suspect that there's a fix that has not been backported to the 0.21 branch. It's going to take some time to find the commit so please keep this open until I find it and get a maintainer to backport it, TIA.

Syres916, 2024-01-26:
@ogrady while this very long job continues, could you please point me to the exact ISO file you downloaded for your distro install so I can create a virtual machine if needed. I'm getting very suspicious that I'm not going to be able to replicate this on Fedora 38. It might be way quicker for me to just build as near as possible the same setup as yours.

ogrady, 2024-01-26:
@Syres916 happy to help, although I am not entirely sure I read you correctly. Are you looking for the ISO to replicate my OS for your VM? If so, then that's [Manjaro x86_64 Plasma Desktop](https://download.manjaro.org/kde/23.1.3/manjaro-kde-23.1.3-240113-linux66.iso). If you are asking for the exact package version of FreeCAD from the mirror, then that is [0.21.2-3](https://archive.archlinux.org/packages/f/freecad/freecad-0.21.2-3-x86_64.pkg.tar.zst).
Let me know if I can provide any additional details to help you pin the issue down.

Syres916, 2024-01-29:
It appears that the root cause is the ArchLinux packagers are getting ahead of themselves when it comes to choosing to upgrade dependencies. I have a way forward to deal with most of the OCC updates to 7.7.2 and 7.8.0. So it's a case of compiling the two OCC libraries locally, applying the updates to 29 FreeCAD C++ files and compiling the two versions of FC. It'll be Thursday I imagine before I get that lot done and if successful create a backport PR.

Syres916, 2024-02-03:
Well, having tried all that I literally haven't gained any understanding as to what is going on. This needs an ArchLinux packaging expert which it's clear that I'm not, sorry but this has taken way too much of my time and I'm having to abandon it.

volker-raschek, 2024-02-03:
Hi, there is a new release of FreeCAD in the staging repository of Arch Linux. It could be possible, that the new release fixed the issue. I've not yet tested the new release.

v0.21.2-4

https://archlinux.org/packages/extra-staging/x86_64/freecad/

volker-raschek, 2024-02-05:
Hi @ogrady  and @Syres916, 
version v0.21.2-4 fixed the issue.

ogrady, 2024-02-06:
Thanks for the quick turnaround and also thank you @Syres916 for digging into the matter.

closed 2024-02-06 by ogrady