FreeCAD/FreeCAD #12094: [Problem] Crash when exiting preferences

pierreporte, 2024-01-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:

1. Open the preferences window.
2. Exit it using OK cancel, close button in title bar, esc key, it doesn’t matter.
3. FreeCAD crashes with the stacktrace below.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35691 (Git) AppImage
Build type: Release
Branch: main
Hash: cc45b58dadae53d7a99be1ed7db25f6c93bb13d5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * TabBar (Disabled)
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

None

### Anything else?

```
FreeCAD 0.22.0, Libs: 0.22.0devR35691 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

NOTICE: Addon "/home/thomas/.local/share/FreeCAD/Mod/TabBar" disabled by presence of ADDON_DISABLED stopfile
Loading Assembly workbench...
NOTICE: Addon "/home/thomas/.local/share/FreeCAD/Mod/TabBar" disabled by presence of ADDON_DISABLED stopfile
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42910) [0x7f4460a42910]
#1  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libpython3.10.so.1.0(PyErr_Fetch+0x12) [0x7f4463151602]
#2  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/./libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15(+0x1653d) [0x7f44628ca53d]
#3  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5(+0x2e9214) [0x7f44610e9214]
#4  0x7f44610e29ff in QObject::destroyed(QObject*) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x2f
#5  0x7f44610e7694 in QObject::~QObject() from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x604
#6  0x7f44610e78d9 in QObject::~QObject() from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x9
#7  0x7f44610ddf70 in QObject::event(QEvent*) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x40
#8  0x7f4461f7035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#9  0x7f44639283f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libFreeCADGui.so+0x99
#10  0x7f44610b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x108
#11  0x7f44610b6c91 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x181
#12  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5(+0x308cb3) [0x7f4461108cb3]
#13  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/./libglib-2.0.so.0(+0x5a417) [0x7f445f711417]
#14  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/./libglib-2.0.so.0(+0x5d3c7) [0x7f445f7143c7]
#15  /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f445f7149be]
#16  0x7f44611083d6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x66
#17  0x7f44610b2bbb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x11b
#18  0x7f44610ba6a2 in QCoreApplication::exec() from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libQt5Core.so.5+0x82
#19  0x7f446388c25d in Gui::Application::runApplication() from /tmp/.mount_FreeCAiYqGlk/usr/bin/../lib/libFreeCADGui.so+0x20ad
#20  /tmp/.mount_FreeCAiYqGlk/usr/bin/freecad(+0x7620) [0x555911e44620]
#21  /lib/x86_64-linux-gnu/libc.so.6(+0x280d0) [0x7f4460a280d0]
#22  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x7f4460a28189]
#23  /tmp/.mount_FreeCAiYqGlk/usr/bin/freecad(+0x7981) [0x555911e44981]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-01-23:
This should already be fixed:
https://forum.freecad.org/viewtopic.php?t=84624

pierreporte, 2024-01-23:
Let’s wait the next weekly in a few days then.

Roy-043, 2024-01-25:
This has been fixed.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

closed 2024-01-25 by Roy-043