FreeCAD/FreeCAD #12099: [Problem] PD Pocketing with DatumPlane followed by PolarPattern corrupts model

hkienle, 2024-01-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You can do a pocketing operation with a datum plane. I am not sure if this is actually an "Easter egg" because it is not document by PD Pocket or PD Plane in the wiki. I like to use it once in a while and it is more elegant than placing a "rough" sketch on the plane and then pocket with the sketch.

In a simple test, pocketing with a plane works as expect, but applying a PolarPattern on the results corrupts the model. (It may be the case that intersecting cuts are causing problems.)

### Pocketing with a datum plane as expected
[image] 

### PolarPattern with 2 instances (no overlap) as expectedf
[image]

### PolarPattern with 6 instances (overlapping cuts) ends up corrupted

[image]

### PolarPattern with 6 instances with SKETCH on datum plane as expected
[image]

### Full version info

```shell
OS: PureOS 10 (Byzantium) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35570 (Git) AppImage
Build type: Release
Branch: main
Hash: 3ba0c3d7958dad891e125e065a41d95faac47c6e
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * workfeature
  * workfeature-macro
  * Assembly4 0.12.4
  * PieMenu 1.1.6
  * Manipulator 1.4.9
```
```

### Subproject(s) affected?

PartDesign

### Anything else?

[bug-pocket_w_datumplane.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14037544/bug-pocket_w_datumplane.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-01-24:
Confirmed
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35649 (Git)
Build type: Release
Branch: main
Hash: 35e09e233c4fadb6194c82164ece25d5043d23fe
Python 3.11.7, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```

Syres916, 2024-01-25:
There may and I stress _may_ be a solution waiting in the wings already, @hkienle did you test with with the 0.19.3 AppImage (no AppImage was produced for 0.19.4) https://github.com/FreeCAD/FreeCAD/releases/tag/0.19.3
I'd guess that it works but possibly a little slow, if so then that points to a reasonably quick solution.

hkienle, 2024-01-25:
> did you test with with the 0.19.3

I did test with 0.19.2 (because I already have it) and the model is still corrupted. :(

```
OS: PureOS 10 (Byzantium) (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

```
[image]

If it helps, I can download and try 0.19.3.

Syres916, 2024-01-25:
> If it helps, I can download and try 0.19.3.

Please do, as that AppImage uses quite a few updated libraries including the OCC but I'm now less hopeful that the solution I had in mind is going to work.

hkienle, 2024-01-25:
Same result with 0.19.3, model is corrupted.

```
OS: PureOS 10 (Byzantium) (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0f9259c)
Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

```

hkienle, 2025-02-09:
Bug is not resolved for latest flatpak (I cannot run AppImage)

```
OS: KDE Flatpak runtime (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39935 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 49b304ae55)
Hash: 49b304ae558553f9acd5872ccd30673bbd60bd19
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: 
  * Gridfinity 0.9.1
```

hkienle, 2025-09-21:
Bug not resolved yet.

```
OS: PureOS 10 (Byzantium) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/02 21:58:19
Build type: Release
Branch: main
Hash: 04e085cba3b43d1e09aedd4445824a6eb46e45b6
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.6656/1
Installed mods: 
  * Assembly4 0.50.7

```

FEA-eng, 2026-10-13:
With the new core datums, I can't apply a pocket directly to the datum plane.

[bug-pocket_w_datumplane_mod.FCStd.zip](https://github.com/user-attachments/files/33370397/bug-pocket_w_datumplane_mod.FCStd.zip)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```