FreeCAD/FreeCAD #12101: [Problem] Unpacking error with Windows portable (.7z) - t64-arm.exe	FAILED

heng7, 2024-01-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The FreeCAD Windows 64-bit portable (.7z) had an unpacking error.
Ref: https://github.com/FreeCAD/FreeCAD/releases/download/0.21.2/FreeCAD-0.21.2-Windows-x86_64.7z

Package testing error message:
Testing FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\Lib\site-packages\pip\_vendor\distlib\t64-arm.exe	FAILED
Error: Unknown compression method for file "FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\Lib\site-packages\pip\_vendor\distlib\t64-arm.exe"
Irrecoverable Error: Test operation failed.

[package_testing.txt](https://github.com/FreeCAD/FreeCAD/files/14039091/package_testing.txt)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

LarryWoestman, 2024-01-24:
Have you tried a very recent version of 7zip?  There was a change roughly 3 months back that required an updated version of 7zip to unzip.

heng7, 2024-01-25:
Hi. Thank you. You are correct.
1. I was using WinZip v26.0 32-bit from year ~2021. This didn't work.
2. Using the latest 7zip v23.01 software, I was able to extract the files without any issues.

closed 2024-01-25 by heng7