FreeCAD/FreeCAD #12105: [Problem] Sketcher uses triangles as arrows.

MisterMakerNL, 2024-01-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

So I was looking at the new sketcher construction lines and looks really good!
But there was something off.
Something wasn't right in my opinion.
And I think I figured out what the issue is.
It's the arrow, or the triangle that is used as an arrow.
I did some research and found out Freecad is the only Cad software that uses the triangle. 
And I think it looks wrong, not saying it is wrong but it looks wrong. It also has this 90's vibe, like before we had normal arrows we used a triangle. So it looks a bit out dated.
[image]
There are also standards for arrows in Drafting, although not necessary to apply here it is nice and looks professional.
See more here: 
http://caddforyou.blogspot.com/2015/06/lines-and-dimensions.html
Size should be as followed:
[image]
Basically length is 3 times the width.

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-01-24:
I think that this issue could be more than that.

First the actual issue isn’t that it’s a triangle: you actually want the triangle to be longer.

Architects use dimensions but they look completely different: there is no arrow but an oblique tick. Since FreeCAD is also for architects, it would make sense to allow this style. Below is an image from AutoCAD. There could be variants but I’m not an architect.

[image]

PaddleStroke, 2024-01-25:
For information in the coin drawing code it is writen something along 'we draw triangles for now but maybe it can be improve to look better'

pitch-circle, 2024-01-25:
The image posted from cadforyou contradicts itself.

At the top, it states a 3:1 ratio between length and width. Lower down it states the angle of the arrow between 15 and 90 degrees.

FreeCAD currently meets this lower requirement, with a 60 degree included angle. Hence, the problem is mis-stated in the sense that the complaint is that FC uses _equilateral_ triangles as arrowheads.

The image also contradicts 12.3.3.1 (page 124) at https://law.resource.org/pub/in/bis/S01/is.sp.46.2003.pdf which is the Standard that the cadforyou page frequently cites.

We might also consider how wise it is to use a webpage from 2015 that cites an Indian Standard for the future direction of FreeCAD. It would be better to seek out the appropriate ISO or US Standard and see what they say.

Finally, Sketcher is not a technical drawing application. We have TechDraw for that. Sketcher is a means to an end, only for 'internal FreeCAD use'. Hence why does it need to conform to technical drawing standards?

Roy-043, 2024-01-25:
We should not create an AutoCAD dimstyle here. That would be overkill. I have no issue with the current arrow heads. Whatever is decided, please don't make them longer.

pierreporte, 2024-01-25:
@Roy-043 Sounds fine.

MisterMakerNL, 2024-01-25:
> For information in the coin drawing code it is writen something along 'we draw triangles for now but maybe it can be improve to look better'

lol, how old is that, and is that still the issue?

MisterMakerNL, 2024-01-25:
@pitch-circle I say in my post it isn't wrong but it looks wrong. There is no deeper meaning here than this looks kinda ugly lets make it more pretty.

PaddleStroke, 2024-02-01:
There it is : 
>         //Draw a pretty arrowhead (Equilateral) (Eventually could be improved to other shapes?)
I'll make a PR changing it to 1/3 ratio

closed 2024-02-01 by chennes