FreeCAD/FreeCAD #12106: [Problem] There are two commands to create a sketch

pierreporte, 2024-01-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To create a sketch, there is two commands: `PartDesign_NewSketch` and `Sketcher_NewSketch`. They act differently in two ways:

* the dialogs to select the base plane is different,
* `Sketcher_NewSketch` can create a sketch inside an assembly (when using #10753) while `PartDesign_NewSketch` always create a body inside the assembly.

Having different behaviors for what appears to be the same thing is bad UX. There should be only one, or at least the result should be the same.

After discussing with @PaddleStroke, we think that it would be best to deprecate one of the command, preferably `PartDesign_NewSketch`, and improve `Sketcher_NewSketch` to use the attachement panel (which would solve #11133 and #11604). The new command would then appear on Sketcher toolbar, PartDesign toolbar, and Assembly toolbar.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

This is part of the improvement of the assembly workbench. A similar approach could be used for `PartDesign_Point`, `PartDesign_Line`, `PartDesign_Plane`, and `PartDesign_CoordinateSystem`.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-24:
There are two commands for good reason, it's two completely different workflows, some of us hate Part Design and use Sketcher, Part and Draft. If the new Assembly Wb needs to create a Sketch then fine do it it in it's own command don't start wrecking others' use. Yorik raised this on the forum earlier this week that existing users should not be having their workflow impacted.

pierreporte, 2024-01-24:
It won’t break your workflow. `Sketcher_NewSketch` looks like this:

[image]

The new command would look like this:

[image]

You can still select the exact same planes and define an offset value. You don’t lose anything: you gain more options that you can avoid if you don’t like them. Granted, it would be a side panel instead of a dialog.

Syres916, 2024-01-24:
All I can say is you obviously don't use this workflow daily, I never need to attach anything, sketches don't have support, leave it well alone.

pierreporte, 2024-01-24:
You are right that I don’t use your workflow. Would you take some time to describe it so we can find a way to integrate it nicely?

WandererFan, 2024-01-25:
That is one intimidating panel for somebody that just wants to make a sketch on a cardinal plane.

Why not have the simple plane selector on top and drop down with all the attachment stuff below?

hkienle, 2024-01-25:
> @pierreporte 
> You can still select the exact same planes

Maybe there is a better workflow, but what I know: in order to select eg. `YZ_Plane` you have to either:
- type it in  (not always possible because "No reference selected..." keeps overwriting you)
- go back to the tree view and navigate and click on the Origin's YZ_Plane object
- make the Origin's YZ_Plane object visible and then click on it in the 3D view

All this is not exactly intuitive...

pierreporte, 2024-01-25:
@Syres916 Thank you for the explanation. You say that you have sketch on the base planes. This is what attachment is. I think that we actually agree with each other, it’s just that we work very differently. If the naming was more consistent across the program, it would greatly help users understanding each other.

@WandererFan Indeed, the attachment panel could certainly be improved. It has nearly all the features and is very powerful already so it’s “only” an UI/UX issue. It only lacks a way to align the sketch axis in specific directions (datum lines or whatever). It can be done independently. @obelisk79 it can be a good subject to work on.

obelisk79, 2024-01-25:
> @Syres916 Thank you for the explanation. You say that you have sketch on the base planes. This is what attachment is.

I think you are wrong here, there is a difference between alignment and attachment in the sense being described. Now, I'm not sure if attaching to base planes would hurt the workflow described by syres or not, but it's worth understanding the difference. There still may be a way to unify the creation of sketches, but we haven't gotten there yet.

> @WandererFan Indeed, the attachment panel could certainly be improved. It has nearly all the features and is very powerful already so it’s “only” an UI/UX issue. It only lacks a way to align the [...]

I'm visualizing some ideas that are kind of popping into my mind as I read through and think about this, but making that task panel better will take some careful thought and consideration. I could probably work out a concept and then bring it for review to the rest of the DWG on Discord before presenting anything here.

pierreporte, 2024-01-25:
> I think you are wrong here, there is a difference between alignment and attachment in the sense being described. Now, I'm not sure if attaching to base planes would hurt the workflow described by [...]

I am indeed. Sketches don’t have an attachement property with `Sketcher_NewSketeh`.

FlachyJoe, 2024-01-25:
> I'm not sure if attaching to base planes would hurt the workflow described by syres

There is no base plane object in a empty document so nothing to attach a sketch to.

pierreporte, 2024-01-25:
There could be an option to position the sketch in space instead of attaching it to the base plane.

If there is no base planes, the list could contain orientation instead.

There could be a pair of radio buttons, one for attachment and one for positioning, and a different interface below depending on the choice.

pierreporte, 2024-04-22:
@maxwxyz for the UI/UX project.

pierreporte, 2024-05-06:
@FreeCAD/design-working-group Can we come up with a UI that could cover both sketch alignment and sketch alignment? Here is a proposal:

[image]

First section is for choosing if the sketch is attached or only oriented. Attachment is the default.

Second section is for positioning the sketch. It is quite similar to what Catia offers for positioned sketches. The list named “attachment mode” in the current attachment panel is displayed somewhere to help the user (it can be a full section of the sketch creation panel). The list is filtered to show only what can be selected: planes and surfaces for the sketch plane, points for the origin, lines for the orientation. The selected orientation can be made the horizontal or vertical axis of the sketch. Flipping buttons can be used to orient the LCS of the sketch more precisely.

When in orient mode, only the sketch plane can be set. It could even accept selecting a line to set the normal instead of the plane.

Third section is the same as the attachment panel equivalent. It can be made a full section of the sketch creation panel.

krushia, 2024-05-07:
Can Orient be the default if nothing it selected before running the command? This would mimic the current functionality we know and love, per https://github.com/FreeCAD/FreeCAD/issues/12106#issuecomment-1909005669

pierreporte, 2024-05-07:
@krushia Yes, it should be the current behavior if nothing but sketch plane is selected. The text box (or whatever it would be) should then reads “default” or similar.

pierreporte, 2024-05-12:
Here is a more complete UI proposal, with the different options illustrated that depend on user choice.

[image]

hkienle, 2024-05-12:
As a side note, I am not a power user: I don't really have a  clue what orient sketch vs. attach sketch means.

I guess the idea is to pre-select for the user based on the context that the sketcher is invoked with?! But since this choice shows up at the very top, as a user I feel I have to make a (informed) decision before proceeding further down...

pierreporte, 2024-05-12:
Oriented sketches are not attached to geometry. They are just positioned in space. It is useful for the Part workbench.

There is indeed a risk that user may not know the difference and make the wrong choice. Most likely, if they work in Part design or Assembly, they will want to attach sketches, it is why it should be the default option. The default mode could differ depending on the workbench.

Mannshoch, 2024-08-30:
Why is there a hard coded Offset to XYZ? I assume it would be better If user could select something like a plane, a surface or an axis. If you wish xyz, why not select it on the gizmo or enter it from a pull-down menu?

PaddleStroke, 2024-12-13:
@pierreporte I like your mockup, but it looks specific to sketch creation. Which sound redundant with attachment dialog anyway no? Besides for the offset I recall seeing another mockup with a combobox letting you choose 'Local coordinate' or 'global coordinate'

pierreporte, 2024-12-13:
@PaddleStroke The mockup can and should be improved. The main change would be to display the same list of available options (like circle center, etc.) so that when clicking on a Select button, the user can have more choice. The idea behind this different attachment dialog is that there could be different dialogs depending on what is attached. It doesn’t make sense to have the same options for a sketch or a line. It can be one dialog but it changes itself.

pierreporte, 2024-12-13:
A better attachment UI would be closer to current attachment panel:

[image]

The difference would be checkboxes for references. Maximum 3 references (sketch plane, sketch origin, sketch orientation). A set of buttons to flip directions. The attachment mode list would be filtered for each kind of reference to only show what is compatible with a plane, a point, and a direction. Plus the ability to either orient or attach.

macdroid53, 2025-03-17:
Having two different "new sketch" workflows needs to be addressed.

If it means having a preference for "long time" users so be it. Or if it is an advanced button in the dialog. Something.

Not being able to actually "locate" a sketch, other than on base planes in Part Design means extra steps to adjust the location when it is intended to be on other than base planes.

Being only able to use a single axis offset in the Sketcher, again means multiple steps.

Maintaining long time inconsistencies is not the way to make FC better.

Mannshoch, 2025-03-28:
What do you think about actively splitting attachment and Sketches into a creating a datum plane and Sketch.

In Solidworks there is the Feature [Plane](https://help.solidworks.com/2025/german/SolidWorks/sldworks/HIDD_DVE_CREATE_PLANE.htm) ([youtube](https://youtu.be/56karjU0vbE?feature=shared&t=77)) which allows to create constrained Planes on 3D objects. Because of that in sketch, there is no difficult Attachement needed. because you could only set a sketch on planes or planar surfaces.

In FreeCAD in sketcher It could mean that, after selecting the feature, instead of presenting a list of planes to chose, in Graphic view non plane objects get dimmed to show you what is ready to click on it and in the Task a button for creating a Datum Plane appear. (if you click on it the sketcher pause and the datum plane feature start.)

macdroid53, 2025-03-28:
In the latest code you can have the plane and LCS  of the sketch displayed. (Currently the user has to turn this on, but, I think it is the intent to have this automatically show while creating a sketch or using the Mapmode dialog. @kadet1090 ?? )

IMO, this would suffice. The biggest reason so many users currently jump to using superfluous and redundant Datum planes is that they expose their location/plane, where, in the past, using Map mode was like fumbling around in the dark.

pierreporte, 2025-03-28:
@Mannshoch That’s pretty much what my UI proposal is: it explicitly asks for a base plane. Though in the long term I’d like to have something like in Catia that allows you to create on the fly datum by right-clicking on an input field.

macdroid53, 2025-03-29:
Let's not mix datum planes and sketches.

In a large percentage of cases, a sketch (especially if it, as it can now, provide visual feedback of its location)  requires no Datum plane. (I'd argue, especially for a single sketch, the Datum is redundant and superfluous.)

Thovthe, 2026-05-03:
> The biggest reason so many users currently jump to using superfluous and redundant Datum planes is that they expose their location/plane, where, in the past, using Map mode was like fumbling around in the dark.

I'm under the impression that sketching on datum planes is a way to make your model more robust against the Topological Naming Problem as long as they aren't also attached to geometry.

macdroid53, 2026-05-04:
> > This is simply not true. The reasons have been explained previously.
> 
> Whereabouts? I'd be curious to learn more.

It is discussed in this thread. And, in the FreeCAD forums, Discord, Facebook group, etc. Even the wiki about TNP makes a point of noting that the Datum plane, used in its example, does not solve TNP.

It is quite simple though. TNP happens when a referenced entity changes name or goes away altogether. For example, a sketch attached to generated geometry (like a face/edge/vertex of a Pad) and after recompute the geometry has changed name. The sketch is now possibly attached to something completely different or nothing. If the a Datum plane is attached (to the a face/edge/vertex), the Datum looses it's attachment when the name changes. Attaching a sketch to the Datum serves no purpose, the datum is redundant, and the datum is jut as susceptible to TNP.

The most stable is to attach to the base planes and use attachment offsets/map mode to locate it in space. You can do that with a sketch, so the Datum is just redundant.