FreeCAD/FreeCAD #12107: [Problem] Assembly: picking JCS attachement zone is not intuitive

pierreporte, 2024-01-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue was discussed with @PaddleStroke after a test of #10764.

To create an assembly joint, when attaching a JCS to an edge or cylinder or similar, it is divided precisely into thirds: one for one end, the second for the middle, the third for the other end. Depending on the orientation and zoom level, it can be hard to pick exactly what is desired. It should work like sketch lines: you have to click very close to the ends to select the vertex while the line is easily pickable.

### Full version info

```shell
0.22 with PR #10764
Tested width Ondsel ES 2024.1
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-04-22:
@maxwxyz for the DWG project. The current implementation is against common UX principles.

@PaddleStroke Any chance this can land in main before feature freeze?

PaddleStroke, 2024-04-23:
This is a bit hard to implement because we are currently comparing the distance from mouse to vertex and select the closest. So it would need to implement a weight mechanism.
So before doing this I'd like to hear more people's thought on this. To make sure this is what most prefer.

howie-j, 2024-04-24:
I find the current implementation very good, although i have not used the assembly workbench much for actual work yet.

@pierreporte do you have a screencast of what you find difficult?  

Here is three different cylinders with aspect ratio from 1:1 to 1:15. Picking the correct JCS is super easy for all three, even without changing zoom level:
[JCS Cylinder.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/f631415a-5bd5-4118-98e8-af429f9e6c24)

pierreporte, 2024-04-24:
> Picking the correct JCS is super easy for all three, even without changing zoom level:

@howie-j Now try to pick the center of the taller cylinder while only the top is visible. This is a simple case with very few parts scattered in the view. In reality, there is often a lot of parts visible and zooming out may not be practical.

I don’t have a screencast. This is based on professional CAD experience.

pierreporte, 2024-05-13:
@PaddleStroke Wouldn’t it be better to do this similarly to picking geometry? If you’re closer than X px to an edge, the edge is selected, otherwise it’s one of the coincident faces. Other CAD programs work like this.

PaddleStroke, 2024-05-14:
So what you want is that basically selecting the cylindrical face, the only available JCS is the center. So basically stop considering the circular edges. And if the user wants the circles he can still select the circles.
Could you provide an example where current behavior does not work properly so I can check it out to consider the issue ?

And this only for cylindrical surfaces? What about plane surfaces? It seems useful that plane surfaces are offering selecting JCS at its vertexes/edges.

pierreporte, 2024-05-14:
> So what you want is that basically selecting the cylindrical face, the only available JCS is the center. So basically stop considering the circular edges. And if the user wants the circles he can still select the circles.

Yes, exactly.

> Could you provide an example where current behavior does not work properly so I can check it out to consider the issue ?

It’s very situational, but the situation occurs very often. When building an assembly, parts are all over the place and move when new joints are added. It happens quite a lot that a part ends up in interference with one another, making hard to select the central zone of a surface without having to move it, which may not be possible if the joint tool is activated. Allowing the full surfaces to be selected reduces the need to move the parts. It will still be necessary because there is always going to have bad situations, but they will be far less common.

Here is a very simple file showing the problem. A drilled part and a pin that should fit in it. The configuration at opening makes it impossible to select the central section of the pin without moving it first.

[jcs picking zone problem.zip](https://github.com/FreeCAD/FreeCAD/files/15311616/jcs.picking.zone.problem.zip)

> And this only for cylindrical surfaces? What about plane surfaces? It seems useful that plane surfaces are offering selecting JCS at its vertexes/edges.

I think that it should be for everything, not just cylinders. Users should be able to select the geometry they want wherever they click on it, except in the picking distance of edges and vertex.

PaddleStroke, 2024-05-15:
I would argue that when you create a joint between 2 cylinders, it does not really matter if you select the center of the cylinder or the top/bottom. The middle is actually not very meaningful and most of the time you would rather select one of the ends as to match with another cylinder end.
Unless you want to match the center of the cylinder to something, but that doesn't feel like a very common usecase.

pierreporte, 2024-05-15:
A revolute or sliding joint between two cylinder actually work with the axis (same for cones and similar shapes that have an axis). It is natural to select the cylindrical surface as mean to select the axis, and is the common way of doing it in the CAD world. The ends may not exist as pickable geometry, like pins that have domed ends (see below), and they don’t carry the meaning of an axis. Also, there is still the problem when parts are in interference and need to be moved before applying the joint.

I think that one problem of JCS is that they select only a point. There seems to be no distinction between points, lines, axis, and surfaces. Its position and orientation is determined by the picked geometry, but once the two are in place, they are just considered as points.

It’s hard to describe with text how the current implementation feels extremely unnatural.

[image]

pierreporte, 2024-12-28:
@PaddleStroke After making a real world assembly with FreeCAD, I can say the current behavior is extremely uncomfortable. Have you changed your mind about this issue?