FreeCAD/FreeCAD #12108: [Problem]  Editing a property and closing with return key changes the document.

edwilliams16, 2024-01-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Installing a fresh FreeCAD on a new computer - zero customization.
Create empty file.
Part WB create Cube
File|New
Part WB create Cylinder
Cylinder Properties-Radius
3 enter
jumps to Start Page

### Full version info

```shell
OS: macOS High Sierra (10.13)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Anything else?

 See forum thread  [https://forum.freecad.org/viewtopic.php?p=735638#p735638](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

edwilliams16, 2024-02-15:
May  be related to issue #7745

mstahl, 2024-08-21:
This is happening to me constantly still in 0.21.2. The only workaround I have that works is simply not having more than one document open at a time, which obvs makes everything a pain.

[code]
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.89.0
  * freecad.gears 1.0.0
  * A2plus 0.4.61
  * fasteners 0.5.27
  * Assembly4 0.50.15
[/code]

edwilliams16, 2025-02-11:
Fixed in

OS: macOS 14.4.1
Architecture: x86_64
Version: 1.1.0dev.40077 (Git) Conda
Build type: Release
Branch: main
Hash: a85756b1c94c30252f99c7bc497beed5c8dcbf25
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 72/127.5
Installed mods: 
  * MeshRemodel 1.10.22
  * DynamicData 2.62.0
  * OpenTheme 2024.9.1
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.28
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.4.25
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.53

closed 2025-02-11 by edwilliams16