FreeCAD/FreeCAD #12110: [TechDraw] Adding new View can be empty and no error messages

iva2k, 2024-01-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a .STEP file with a design from SolidWorks (I tried file from 2015 and 2021 SW versions), which imports and shows up correctly. Then I create a TechDraw page and create few views (either with "Insert Projection Group" or individually) - Top, Front, Right, and only Top view ever shows, with Front and Right Views never showing up. Some angled views I tried show up.

I tried different ways to import the design, including intermediate translation using "CAD Assistant" to some different formats (STL, IGES), but the same end result.

I also expanded the design hierarchy, and there are 2 parts - plastic enclosure parts for injection molding that seem to be the culprits, all other parts make working Views when done separately (they are quite complex with a lot of sculpted and slanted surfaces).

I also tried Link branch, it has the same problem with this design.

There are no error messages in the Report.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

I can't post the .STEP or FC file (or screenshots) in this issue, but I can PM it to help reproduce.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-01-25:
Can't really diagnose this without a sample.  You can pm me on the forum or to wandererfan at gmail.com

Syres916, 2024-01-25:
> I also expanded the design hierarchy, and there are 2 parts - plastic enclosure parts for injection molding that seem to be the culprits,

Is it possible that these are actually shells and not solids?

In order to show what FreeCAD 'thinks' the feature is, just select one of the 'culprits' and copy the following into the Python Console:

`Gui.Selection.getSelectionEx()[0].Object.Shape`

On my test step assembly I use for benchmarking each feature returns something like:

`<Solid object at 0x55ecc08fda90>`

The Part workbench has the ability to make a solid from a shell  https://wiki.freecad.org/Part_MakeSolid it's not parametric but that's going to be irrelevant in this case.

iva2k, 2024-01-25:
@Syres916 - Thanks! I did this check, and the culprit parts report as solid. 

I also ran a geometry check in Part workbench, and it reported some boolean operation errors, a bunch of these:

```text
Edge - GeomAbs_C0
Face - GeomAbs_C0
Edge - Invalid curve on surface
Face - Invalid curve on surface
```

I'm not sure if those can be the problem, and how to fix them.

@WandererFan - I sent you the zipped .FCStd file. Let me know if you want the .STEP.

iva2k, 2024-02-14:
I suspect it might be due to silently running out of memory. My Windows 10 laptop has 16GB, but I often have massive memory hogs running: Altium, Brave with many tabs, and FreeCAD manages to quickly come on top of memory use quickly growing when using TechDraw.

I just worked with another set of .STEP files (one includes Raspberry Pi Zero 2 W downloaded from grabcad). Raspberry Pi did show all the views at first, but a bit later stopped showing some of the views without any errors in the report. So I restarted FreeCAD and closely monitored Windows task manager. The memory started at about 300MB and was going up and up while it tried to display the 2D views, and when total system memory grew to over 90%, views stopped rendering (and at some experiments it got to 100%). When closing all files in FreeCAD, memory does not go down, it only ever goes up. 

So there are massive memory leaks in TechDraw (and maybe elsewhere) in FreeCAD 0.21.2 on Windows. And amounts of memory it takes are exorbitant - easily reaches 2GB (as much as OS is willing to give it).

With my Software Engineer hat on, I can observe few problems:

1. Not reporting when memory allocations fail, 
2. Not aborting procedures when memory allocation fails and can lead to freezing up.
3. Not freeing big chunks of memory. Considering Python, there must be references floating around and not deleted, preventing garbage collection,

WandererFan, 2024-03-07:
"So there are massive memory leaks in TechDraw (and maybe elsewhere) in FreeCAD 0.21.2 on Windows. And amounts of memory it takes are exorbitant - easily reaches 2GB (as much as OS is willing to give it)."

I think you'll find this high memory usage when importing step files even when TechDraw is not involved.  I'm not saying that there are no leaks in TechDraw, just that this does not seem to be the case here.

WandererFan, 2024-03-07:
If you get the empty box for a view, it means we did not get any geometry back from the HLR process.  We should display a message somewhere when this happens.

WandererFan, 2024-03-07:
Do you still have the original step file?   I have another issue with step files and need some complicated test files.

iva2k, 2024-03-10:
@WandererFan - I emailed you the .STEP file for that project.

PaddleStroke, 2024-03-11:
Have you tried to wait for a long time?
In a recent test, I had this 'empty view' thing happening when trying to make a view of 66mb step file. Nothing showed and the view seemed empty. But then after 10 minutes the view appeared.
So it may just be that the process is done in a separate thread and takes a long time with no indication something is going on.

iva2k, 2024-03-11:
In one of first tries I gave it 2-3 hours, still nothing. As I said, it feels like it works hard internally and allocates gobs of memory, until it hits a wall and gets no more memory, then the background process hangs and spins forever. Killing it releases the CPU and frees the memory.

I don't buy why TechDraw needs such big amounts of memory. Regular FreeCAD 3D view of the same design with bunch of STEP files displays the views rendered and shaded and lets it spin and zoom and pan in real time, without burning all the memory in the machine. Perhaps the problem is that TechDraw implements the same 3D to 2D in pure python without GPU and has very little control over its memory requirements and not much of error handling. Just a guess.

luzpaz, 2025-06-10:
bumping for discussion

FEA-eng, 2025-10-20:
@WandererFan Should this issue stay open? It seems you got the example file from the OP.

FEA-eng, 2026-04-21:
See [this](https://github.com/FreeCAD/FreeCAD/issues/12110#issuecomment-3420773410)

luzpaz, 2026-09-11:
@WandererFan ping

WandererFan, 2026-09-12:
The file is slow to draw, but it takes minutes here, not hours.  Even checkGeometry from Part is slow with this stp file.  

[image]

```
OS: Linux Mint 22.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 26.3.0dev.48516 +1 (Git)
Build date: 2026/09/09 14:22:03
Build type: Debug
Branch: LockedViewPositionOnLoad
Hash: 4f9963a87af18eabc919928e809c96f3396bc115
Python 3.12.3, Qt 6.11.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 6.11.0
shiboken 6.11.0, SMESH 7.7.1.0, xerces-c 3.2.4, Clipper2 , IfcOpenShell 0.8.5, OCC 7.9.3
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/129.405/1.09375
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 25.2.8-0ubuntu0.24.04.2/Intel/Mesa Intel(R) UHD Graphics 620 (WHL GT2)
```

WandererFan, 2026-09-12:
Redrawing the page peaks at 1.2Gb of memory on my machine, but that is on linux, not Win. 

If you see an empty border (harder to see with "view frames" set to auto), then there will also be a message in the report view - "DVP::execute - View - Source shape is Null.".

If you don't see a border at all, then the most likely cause is that OCCT HLR is still working.  You can turn on "Report progress" in the Advanced preference tab to see what view HLR is working on and whether it is finding hidden lines or finding faces.