FreeCAD/FreeCAD #12115: [Feature Request] Suppressing FEM features

FEA-eng, 2024-01-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working with FEM, it's often useful to temporarily remove some features (like loads or boundary conditions) from the analysis  to test something. In FreeCAD FEM, features can only be deleted and manually redefined which is not very handy in such cases. Especially since the newly added features go below the results in the tree, causing some mess. It would be great to have an option to suppress FEM features until they are resumed for further use. Something like this for non-FEM features is a subject of this PR: #12096 but the logic here is different so it would require a separate solution.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35691 (Git)
Build type: Release
Branch: main
Hash: cc45b58dadae53d7a99be1ed7db25f6c93bb13d5
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Dutchtallman, 2024-01-31:
Suppressing features for FEM purposes might not be limited to the FEM workbench. Sometimes, you want to de-feature your CAD model as well (suppress small holes, fillets, roundings), so that your element mesh becomes smaller and the calculation runs faster... Perhaps `suppressing features` should be a more generic capability in FreeCAD

FlachyJoe, 2024-02-12:
#12359 implements a suppressed property for FEM Constraints.
What are the other FEM objects to make suppressible ?

maxwxyz, 2024-09-21:
What is left here?

FEA-eng, 2024-09-21:
> What is left here?

Suppressing FEM objects other than "constraints" - meshes, materials, element geometry definitions.

PR #18636 (2025-01-06): Fem: Add suppressible extension to objects - fixes #12115
Fixes #12115.
@FEA-eng
closed 2025-01-06 by yorikvanhavre