FreeCAD/FreeCAD #12119: Unable to create groove that intersects axis of rotation

lucian151, 2024-01-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is helpful for a user to able to revolve cut (Groove) a circular sketch entity about a bisecting Datum Line or Sketch Axis, like in the circumstance shown below:
 [image]
[image]

However when someone attempts to do this, an error window appears stating `Revolve axis intersects the sketch`. People coming from many other CAD programs (myself included) may expect this behavior to work just fine without error. FreeCAD should strongly consider making this workflow allowable.

[image]

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * freecad.gears 1.0.0
  * OpenDark 2023.10.2
  * Render 2023.8.18
  * Glass (Disabled)
  * QuickMeasure 2022.10.28
  * A2plus 0.4.60i
  * Manipulator 1.5.0
  * fasteners 0.4.54
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.6.0
  * Assembly4 0.12.5
  * Assembly3 0.11.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

Attached is the file I used to create the screenshots shown above: [mobo-top-cover-alt-dimples.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/14057221/mobo-top-cover-alt-dimples.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-28:
I’ve never encountered such behavior. It could lead to unexpected results if users are unaware of it. Is there any CAD program doing this, except FreeCAD derivatives?

FEA-eng, 2024-08-28:
> Is there any CAD program doing this, except FreeCAD derivatives?

3DExperience (CATIA v6) throws an error about the intersection between the axis and profile too.

Mannshoch, 2024-08-30:
In solidworks, in such cases you could select the sketch-regions which should be used for the cut.
https://help.solidworks.com/2024/english/solidworks/sldworks/c_revolves_folder.htm

    create a sketch -> start feature -> auto choose the only available construction line.
    create a sketch -> start a feature -> if more than one construction line, or none is available you have to select a construction line, contour or an edge
    create a sketch -> select a straight line/contour (within the sketch) -> start feature  -> the feature auto choose your selection if it's the only selection.

(in Solidworks most feature support pre-selecting, if a feature need selectable things e.g. a plane and a line as requirement, in most cases you could select required references before starting the feature)

pierreporte, 2024-08-30:
I propose this:

1. Revolving a sketch that have geometry on both side of the axis is forbidden.
2. It is allowed to select a sub-secion of the sketch for all operations, including grooves, as requested in #12820.
3. The revolution axis cuts geometry that intersects with it (when groove function is called, not in reality).

Then an error will be displayed to the user if the selection is on both sides of the revolution axis, be it the full sketch or a subset (one or more closed contours).

maxwxyz, 2026-08-16:
@pieterhijma FYI