FreeCAD/FreeCAD #12120: [Problem] Managing Custom Data Elements

pieterhijma, 2024-01-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are multiple competing solutions for users to create and maintain custom data elements.  There should be a single core feature that is universally used. It should be accessible from both Python and C++ workbenches as well as providing strong support via the API.

The competing solutions are Spreadsheets, Variables in the Assembly 4 workbench, PropertyBags in the Path workbench, and the Dynamic Data workbench.  Because there is no standardized solution, there is no common GUI support and because of the separation, there is much potential to improve the general GUI support and workflow of managing data.

Unifying managing custom data elements also gives opportunities to standardize related concepts such as configuration management (both Spreadsheet and Dynamic Data provide solutions) and variant links (a solution with Spreadsheet configuration tables exists and an experimental implementation in the Assembly 4 (that only works in Assembly 4)).

Finally, having a core solution for managing data paves the way for improving modularity problems in FreeCAD designs where designs often rely on a single file with globally shared data.  In principle, this works well, but the various files with designs are not modular any longer because they rely on the single file with globally shared data that may have different global values not relevant for other projects.

Please consider this as an overarching issue.  I will create smaller issues and PRs addressing issues one by one.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35729 (Git)
Build type: Release
Branch: main
Hash: 8d88cc7e79e8e3bc9090ab8f1b244bc9a616e940
Python 3.11.6, Qt 5.15.12, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Ondsel-Lens 2024.1.22.01
  * backup
  * BIM 2021.12.0
  * Assembly3 0.12.2
  * fasteners 0.5.0
  * Assembly4 0.50.6
  * osh-autodoc-workbench
  * DynamicData 2.60.0
```
```

### Subproject(s) affected?

Core

### Anything else?

### Terms

Because many use different terms for similar things, let's define some terms for the context of this issue and related issues and pull requests.

- **Custom Data Elements**: A data element defined by the user for use in models. Examples are spreadsheet cells (that may or may not be aliased) and properties in document objects.
- **Configurations**: Geometry that can be configured to change.  For example, consider a model for screws that can be configured to be an M8 or M10 screw, reusing all the underlying geometry, but changing the data to reflect the requested configuration.
- **Variant Link**: a link to geometry that can be varied. For example, given the geometry above that can be configured, it is possible to have two copies of the geometry, one for an M8 and one for an M10 screw. Instead of copying, we could also create a special link that allows us to reuse the same geometry but vary only the configurations.

### Relevant Sources of Information

#### Ondsel blog post

- [Ondsel Blog post](https://ondsel.com/blog/build-system-for-custom-data-elements/)
- [Ondsel issue](https://github.com/Ondsel-Development/FreeCAD/issues/5)
- [Initial PR](https://github.com/FreeCAD/FreeCAD/pull/10653)

#### Forum posts

- [Where to store variables](https://forum.freecad.org/viewtopic.php?t=84118)
- [Comparison Fusion 360 Configurations](https://forum.freecad.org/viewtopic.php?p=719433)
- [Changing parameters (referring to ASM4 variant)](https://forum.freecad.org/viewtopic.php?p=714337)
- [ASM4 Variant links](https://forum.freecad.org/viewtopic.php?p=710284)
  - [Specific post on master variables](https://forum.freecad.org/viewtopic.php?p=710284#p710284)
- [Workflow for variants](https://forum.freecad.org/viewtopic.php?p=700016)
- [Variant Link Assembly 4](https://forum.freecad.org/viewtopic.php?t=62767)
- [Configuration tables](https://forum.freecad.org/viewtopic.php?p=662447)
- [Possibilities of the 'Part'](https://forum.freecad.org/viewtopic.php?t=84334)

#### Wiki

- [Variant Link](https://wiki.freecad.org/Variant_link)
- [Subshapebinder](https://wiki.freecad.org/PartDesign_SubShapeBinder)
- [Spreadsheet Configuration tables](https://wiki.freecad.org/Spreadsheet_Workbench#Configuration_tables)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

joshualibrarian, 2024-01-25:
I think this is really important.  As I lay out in that _Possibilities of the 'Part'_ post, I would like to see the already-existing custom object parameters be leaned into, since they are very flexible and already there.  As I said, I also like the structure of using enclosing Part (or Body) objects to hold parameters for their children, helping to create more encapsulated models.  I think this opens naturally to the sorta "template" pattern I suggest in my post, which I think would be a much more intuitive pattern than the current spreadsheet setup, making variant links much more accessible and leaning into the fantastic foundation of Links.

Zolko-123, 2024-01-26:
> There should be a single core feature that is universally used

why ?

kadet1090, 2024-01-26:
> > There should be a single core feature that is universally used
> 
> why ?

To not reinvent the wheel over and over again. This should not mean however that for example Assembly4 should not be able to extend this system to fulfill specific needs - the data should be however available in coherent and interoperable manner via common interface.

pieterhijma, 2024-01-26:
> > There should be a single core feature that is universally used
> 
> why ?

My apologies, I tried to make the reasons clear in the description. Let me list them one by one and add some details:
- There are multiple solutions to the same problem, sharing functionality but not code, so satisfying [WET](https://en.wikipedia.org/wiki/Don't_repeat_yourself#WET).
-  Because there is no standardized solution, there is no common GUI support and no common workflow.
- Different files use different strategies for managing data, requiring users to first understand which data management strategy is used.
- It gives opportunities to standardize related concepts such as configuration management and variant links.
- It paves the way for solving modularity problems. Modularity problems often originate from the relations between geometry objects, typically expressed in references to data of those objects. If there are multiple solutions to manage the data, the modularity problem is much harder to solve.

pieterhijma, 2024-01-26:
#12135 is a draft PR that introduces a VarSet. Since this is only a small part of the solution I have in mind, the PR describes all future plans for context.

Roy-043, 2024-01-27:
One of the current problems is that any change to a spreadsheet (even adding a value in an empty cell) triggers a full recompute of the model. I believe DD has the same issue. A new solution should ideally also address that problem.

Zolko-123, 2024-01-27:
> To not reinvent the wheel over and over again

then why do you do it with this ? 

> There are multiple solutions to the same problem

And so you're adding **another** solution to the same problem ? Have you ever read [xkcd](https://xkcd.com/927) ?

[image]

> One of the current problems is that any change to a spreadsheet (even adding a value in an empty cell) triggers a full recompute of the model

that's more of an annoyance than a problem, but I agree that it can be improved. The obvious solution is to have a check-box or an option that sets "auto-update" (or similar), and when set to false then you can make all modifications without any intermediate recompute, and make the full recompute at the end when all variables are changed.

joshualibrarian, 2024-01-28:
@pieterhijma In my latest model, I'm still using Spreadsheet for configuration tables, but to just plain store variables, I'm using plain old `Group` objects, with custom properties.  I can reference them from anywhere, whether I put them in the `Part` or not, and I can nest them cleanly and reference them with a dot syntax.  This replaces the manual grouping with underscores in actual Spreadsheet cells, which are really just properties in the Object anyway, just with cell coordinates.

The other big advantage to doing it this way is I can have separate objects for different sets of variables (I have a lot of variables in my models), this way when I change a value in one group, ALL of the groups don't need to recompute.

pieterhijma, 2024-01-28:
> One of the current problems is that any change to a spreadsheet (even adding a value in an empty cell) triggers a full recompute of the model. I believe DD has the same issue. A new solution should ideally also address that problem.

Thank you for bringing that up. I would agree that that would be ideal too. It is definitely related, and my intuition about that problem is that it is mostly a problem of lack of modularity. So, I hope that a solution to modularity problems will also solve recomputation problem.

pieterhijma, 2024-01-28:
> > To not reinvent the wheel over and over again
> 
> then why do you do it with this ?

I feel this is directed to me, yet the quote is not mine. The reasons why I bring up this issue are in [this post](https://github.com/FreeCAD/FreeCAD/issues/12120#issuecomment-1912026193).

I try to keep the description of the problem (this issue) separate from potential solutions. Regarding one potential solution I proposed in PR #12135: I don't think that is reinventing the wheel, it is more consolidating, recognizing the usefulness of it from various addons, bring it to core, and add more functionality to it.

Jolbas, 2024-01-28:
> One of the current problems is that any change to a spreadsheet (even adding a value in an empty cell) triggers a full recompute of the model. I believe DD has the same issue. A new solution should ideally also address that problem.

A possible workaround is to right click the document in the tree view and check "Skip recompute" and then uncheck it or hit Command+R when you are done editing.

Roy-043, 2024-01-29:
Skipping recomputes is indeed possible but does come with a big disadvantage:
https://wiki.freecad.org/Spreadsheet_Workbench#Complex_models_and_recomputes

Jolbas, 2024-01-29:
> Skipping recomputes is indeed possible but does come with a big disadvantage: https://wiki.freecad.org/Spreadsheet_Workbench#Complex_models_and_recomputes

New or changed raw values is shown but not new calculated values. The same happens with properties in property view. But that's expected, at least by me.

Roy-043, 2024-02-03:
Related issue: ##5948.

pieterhijma, 2024-02-21:
Dear all, I've created a video to explain the problem better. This is Part 1 that discusses the problem. PR #12532 is a draft that shows a potential solution in terms of VarSets. That PR contains Part 2 of the video that shows the same example in Part 1 but then with VarSets.

- [Custom Data Elements, Variants, Modularity - Part 1](https://github.com/FreeCAD/FreeCAD/assets/5336482/0c425b24-2874-411f-8512-e73cbdc4d1d6)

maxwxyz, 2024-02-24:
@pieterhijma would it be possible to expand the parameters in the tree view to show all parameters with their **names and values** below (right in the tree)? Like you can expand a part design body and see all features of it.

I like this feature in CATIA as you quickly get an overview and can change or copy only one parameter:
[image]

[image]

pieterhijma, 2024-02-29:
> @pieterhijma would it be possible to expand the parameters in the tree view to show all parameters with their **names and values** below (right in the tree)? Like you can expand a part design body [...]

@maxwxyz Sorry for the late reaction. I definitely see the usefulness of that. In principle everything is possible, but in this case it would come with a very high cost because of FreeCAD's design. FreeCAD has a strong separation between `DocumentObject` (for example a Part Design Body) and `Property` where document objects have properties. The tree shows only document objects. 

To acquire the above, there are a couple of solutions:
1) Allow properties to show up in the tree: Drawback is that document objects typically have many properties which would make the document tree unusable. You could filter on certain properties, but which properties should you filter then? For this solution, it will always be the case that properties are muddled with document objects, making the semantics of both properties and document objects unclear.
2) Create a `DocumentObject` `Parameter` that only maintains one property that is the parameter. This would be possible, but FreeCAD is not designed for this, so this will have probably all kinds of complications, for example also performance complications. Additionally, it diminishes the value of properties.

So, both solutions are not very satisfactory and solutions inbetween would still affect the separation between document objects and properties. I would suggest to not go this route and find a different way that better fits FreeCAD's design to:
1) make an overview of parameters and their values and
2) make sure that values are easy to copy.

Thanks for mentioning this. I'll take it into account when I will work on GUI support for VarSets.

maxwxyz, 2024-02-29:
Thanks for clarification. Would it be possible to show all parameters (name + value) under the parameter set DocumentObject (expandable in the tree, as you could expand a feature and see the sketch). Or maybe choose only which parameters could be exposed to the tree under the parameter set (maybe default all on). A double click in the tree on a child parameter of the parameter set then could directly open a dialog to edit this parameter.

Also, it seems like the parameters are all without a unit. Would it make sense to attach a unit (optionally) to a parameter? Mostly this will be mm but what about an angle. Or if my system is set to imperial units and I would copy over a parameter set from a metric document?

maxwxyz, 2024-03-09:
@pieterhijma there were two mentions at the FreeCAD Day 2024 Complaint Session, I think they could be addressed with this FR, please let me know otherwise.

**There is no way to globally set a reference dimension**
Data management - I don't like the spreadsheet.
It would be nice to have a window where it lists so you have sketch on lists all the dimensions and if you want to assign that as a parameter you just give it a name. You can do that in sketch mode, but you can't oversee all of the dimensions. So if you want to work within multiple sketches, you don't have a window, which overlooks all.

**Parametric design its impossible without a spreadsheet**
A lot of things are more difficult to do without a spreadsheet than with one. 

**People who are working on data management features don’t know what users want.**
Maybe this is more a call to complain about data management, so I have complaints about data management working with the spreadsheets. The complaint is people who are currently working on data management features don't know what users want and don't know how to get that information.

pieterhijma, 2024-03-11:
@maxwxyz, that's funny. I was there at the complaint session. I'm pretty sure I was responsible for the first one. The second one, I don't remember. It is possible without a spreadsheet but the typical way is spreadsheets. Variable Sets can change this. I don't remember the third complaint. So, I'm working on it but largely based on my own frustrations as a user. So, I hope that if I improve it for me, it improves for others as well. The fourth one is probably coming from me.

pieterhijma, 2024-03-11:
> @pieterhijma Thanks for clarification. Would it be possible to show all parameters (name + value) under the parameter set DocumentObject (expandable in the tree, as you could expand a feature and see the sketch). Or maybe choose only which parameters could be exposed to the tree under the parameter set (maybe default all on). A double click in the tree on a child parameter of the parameter set then could directly open a dialog to edit this parameter.
> 
> Also, it seems like the parameters are all without a unit. Would it make sense to attach a unit (optionally) to a parameter? Mostly this will be mm but what about an angle. Or if my system is set to imperial units and I would copy over a parameter set from a metric document?

I would prefer not to have that functionality as it blurs the boundaries between document object and properties. If you want this functionality, I propose to create a separate issue for that, but perhaps it is better to wait after Variable Sets have been implemented with improved GUI support. It may be the case that the problem is not that important anymore at that stage.

For the second point, I believe you are referring to the video? I was just lazy and didn't use units, so I used PropertyFloats. It's perfectly possible to use PropertyLengths giving you units.

pieterhijma, 2024-03-11:
> @pieterhijma In my latest model, I'm still using Spreadsheet for configuration tables, but to just plain store variables, I'm using plain old `Group` objects, with custom properties. I can reference [...]

Sorry, I missed this comment. This sounds similar to having a VarSet object, but I don't quite get from this comment how configuration tables works.

maxwxyz, 2024-07-21:
@pieterhijma can this issue be closed after VarSet support was merged?

pieterhijma, 2024-07-23:
In my opinion managing custom data elements is not solved. Although VarSet support has been merged I wouldn't say that configurations and variants are solved. I would advise to leave this open as an overarching issue.

3x380V, 2024-09-29:
>     * There are multiple solutions to the same problem, sharing functionality but not code, so satisfying [WET](https://en.wikipedia.org/wiki/Don't_repeat_yourself#WET).
>     * Because there is no standardized solution, there is no common GUI support and no common workflow.

Is there a plan to deal with that? Without further consolidation [xkcd](https://xkcd.com/927) episode linked above could add yet another item to its ever growing list of examples.

So we have:
[image]
_Add Property_ is basically the very same, except the ordering:
[image]
`PropertyBag`:
[image]
And _Asm4_variablesCmd_
[image]

With A4 being external workbench the rest is in-tree inconsistency.

furgo16, 2024-09-29:
Reported as https://github.com/FreeCAD/FreeCAD/issues/14633, but seemingly not currently being worked on.

3x380V, 2024-09-29:
@furgo16, thank you. I was not aware about that. Sadly, it seems all this will be released as 1.0 :-/

Zolko-123, 2024-09-29:
> **Problem description**
> There are multiple competing solutions for users to create and maintain

1) there is no real problem here, since there are actually several solutions to a common problem. So the real problem is to get rid of false negatives

2) if we really accept this multi-solutions as a problem, the entirety of FreeCAD must be reported here. Therefore, I propose to close this as "_fontfix_"

3) now, getting to the root of the problem, the real problem here is inconsistency: several tools which do the same thing but differently ... as per the user interface. Therefore, the solution is trivial, as in any other project : UI guidelines. As maintainer of Asm4, I don't care where the CANCEL and OK buttons are, it's 1/2h of coding, but what is it that [I'm expected to do](https://www.youtube.com/watch?v=wbkI0dc2iRg) ? There are no guidelines ?

3x380V, 2024-09-30:
@benj5378, I thought it might be useful to have multiple Asm4 `Variables` in single document - instance per `Part` and one living in `Assembly`. And while touching the code, I was about to give `VarSets` a try. Now I suggest just wait until at least core FreeCAD features gets polished, in another words until `VarSets` reaches the goals i was created in the first place. And btw, I second this https://github.com/FreeCAD/FreeCAD/pull/12135#issuecomment-2199314774. Either `Properties` or `Variables`.
The only annoyance is it will be released with 1.0 creating more user confusion.

3x380V, 2024-09-30:
@Zolko-123, for a _variantLink_ and yes, I ended with `DynamicData` workbench too. Anyway, this is off topic here. It is only useful to point out `VarSet` will not currently change anything here, see #12532.

gbroques, 2024-12-29:
Thank you for formalizing, elucidating, and pushing this along @pieterhijma.

I'm currently working on an open-source locally manufactured small wind turbine in FreeCAD.

The generator / alternator of the wind turbine can be reused in small hydro applications as well.

However, there's currently no good way to decouple the alternator from the wind turbine model since it's dependent upon a global spreadsheet with a bunch of data elements specific to wind.

You'd have to copy-paste the Alternator files, global spreadsheet, and then remove all the irrelevant data elements.

I'm cautious about depending upon any external addons due to long-term maintainability of the project, so bringing this functionality to core is important to me.

Anyways, wishing you and the rest of the FreeCAD team the best on your continued development, [variant links proposal](https://github.com/FreeCAD/FPA-grant-proposals/issues/10), and looking forward to trying this feature out once it's a bit more stable.

pieterhijma, 2025-01-09:
Hi @gbroques, So cool to hear about your case! I'm glad to hear that someone runs into the same issues as I do all the time. I created a couple of videos on this lack of modularity above. Let's hope we can make progress in this respect!

JacobTheEldest, 2025-06-10:
This issue still exists

FEA-eng, 2025-08-20:
I guess it's still relevant

FEA-eng, 2025-10-20:
@pieterhijma I assume it's still relevant, right?

pieterhijma, 2025-10-20:
Yes, it is indeed.