FreeCAD/FreeCAD #12124: [Problem] UserEditModes not passed to App::GeometryPython objects

Roy-043, 2024-01-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

UserEditModes are not passed to App::GeometryPython objects.

Steps to reproduce:
1. Edit Mod/Arch/ArchBuildingPart.py: Add `print(mode)` as the 1st line in the `setEdit` function.
2. Restart.
3. Switch to the Arch WB.
4. Create an [Arch_BuildingPart](https://wiki.freecad.org/Arch_BuildingPart).
5. Change [UserEditMode](https://wiki.freecad.org/Std_UserEditMode) to Cutting.
6. Double-click the BuildingPart in the Tree view.
7. Result: mode should be 2, but is zero.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-01:
@yorikvanhavre @Roy-043 is this still relevant for 1.0?

luzpaz, 2024-07-11:
@yorikvanhavre is there anyone you suggest delegating this to ?

yorikvanhavre, 2024-07-15:
Not really, it's definitely a core problem, but I'm not exactly sure where the underlying issue lies... Maybe @wwmayer would have an idea?

wwmayer, 2024-07-15:
> Change [UserEditMode](https://wiki.freecad.org/Std_UserEditMode) to Cutting.

There was never any guarantee that a view provider must implement all the hard-coded edit modes. Especially Cutting doesn't make sense for part objects.

Roy-043, 2024-07-17:
The specific edit mode in the example is just that: an example to demonstrate a bug.

luzpaz, 2024-10-27:
GIven that, where are we at here ?

Roy-043, 2025-02-17:
This is still an issue.

FEA-eng, 2026-09-30:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/bc8458b4e5a150e91c18c5f2f4cf370ac9c635c6

Double-clicking a geometry object passes the selected edit mode through instead of always using 0. Could be picked up for a PR.