FreeCAD/FreeCAD #12125: Adding App::GroupExtensionPython breaks Std_TransformManip for App::GeometryPython objects

Roy-043, 2024-01-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Adding the `App::GroupExtensionPython` extension breaks [Std_TransformManip](https://wiki.freecad.org/Std_TransformManip) for `App::GeometryPython` objects: the transform widget always appears at the global origin ignoring the Placement of the object.

Steps to reproduce:
1. Create the object with: `obj = FreeCAD.ActiveDocument.addObject("App::GeometryPython","BuildingPart")`
2. Make the [axis cross](https://wiki.freecad.org/Std_AxisCross) visible.
3. Move the object along the X axis with Std_TransformManip.
4. Exit the command.
5. Repeat the previous 2 steps.
6. OK so far.
7. Add the extension with: `obj.addExtension('App::GroupExtensionPython')`
8. Call Std_TransformManip.
9. Notice that the widget now appears at the origin.
10. This reset occurs whenever you call the command again.

### Full version info

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35720 (Git)
Build type: Release
Branch: main
Hash: 00cd50bbf49c810555899efe112ec7595c76cd3a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-03-21:
It was working fine until v0.18 and is broken since v0.19.

The function to blame is DocumentObject::getSubObject. If an object has no extension then the transformation matrix is computed from the placement property.

But if the object has an extension then DocumentObject::getSubObject returns immediately after calling extensionGetSubObject of the first extension without setting the transformation matrix afterwards.

See https://github.com/FreeCAD/FreeCAD/blob/main/src/App/DocumentObject.cpp#L847

Roy-043, 2025-02-13:
This issue has been fixed.
```
OS: Windows 8 build 9600
Architecture: x86_64
Version: 1.1.0dev.40176 (Git) Conda
Build type: Release
Branch: main
Hash: 2745f436026b6f1fb84dbaaf89ea22b5fb4c2295
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell /.0.0, OCC 7.8.1
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/86.4394
Installed mods:
```

closed 2025-02-13 by Roy-043