FreeCAD/FreeCAD #12141: [Problem] Assembly: displayed names are not customizable

pierreporte, 2024-01-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the assembly workbench, the displayed name of parts and assembly is fixed. It must be modified manually for each and it doesn’t update when inserted parts are renamed in their own files.

In the industry when using part numbers and where several parts can have the same name (because it’s the part number that counts and you quickly run out of name ideas) so the tree often displays both the part number, the name and a few other attributes depending on policy and user preference: version, revision, translated name, and of course instance number or name (see #12139).

Currently, there aren’t usable part attributes (see #12136). Then it is resolved, there should be a program settings that allow to build the display of parts and assemblies. It can be done in a similar fashion as date codes like %Y, %mmm, etc. At the end of the issue is an example from Catia. This program doesn’t have very explicit naming so I propose full-text like {part number} or {name} using the names defined in the attributes list presented in #12136.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]

More details on how it works with attributes, including custom ones: http://catiadoc.free.fr/online/bascupst_C2/bascupst0300.htm

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-01-26:
This also apply to parts opened alone.

pierreporte, 2024-02-26:
This is about the assembly workbench because I don’t want to make assumption about the arch/BIM part of FreeCAD, though it certainly needs something similar.

Let me try to explain what should be happening.

You have a file called bolt.fcstd. This file has a Part object in it. The Part object has attributes (see #12136 and #12120). Since we are in mechanical design, it is likely that there will be at least these attributes: `part_name` and `part_number`. The file name is itself some sort of attribute: `file_name`. When a part is inserted into an assembly, it gets a new attribute (stored in the assembly container) for the instance number `instance_number`.

Say the attributes are empty. When bolt.fcstd is opened, the Part container text in the tree view is the file name: “bolt”. If the attributes are set, it could display something like “123456 - HEX SCREW M8x20”. When the part is inserted into an assembly, the same Part container could have the following label: “123456 - HEX SCREW M8x20 [3]”.

In the settings you will have two options: one to set the tree view text when the component is a root in the current view (like bolt.fcstd opened alone or an assembly), and one for the components that are children of the root (or children of children) in the current view. Something like this:

* root text: `{part_number} - {part_name}`;
* child text: `{part_number} - {part_name} [{instance_number}]`.

It is the user that chooses what is displayed. Of course there is a default settings. You could make FreeCAD to always display the file name or never display the part number. If you define other attributes, a translated name for instance, you can edit the strings to make it appear.

Now you’ve determined that the part has the wrong name. You edit the `part_name` attribute to “HEX SCREW M8x20 A4-80 PASSIVATED”. At this moment, the text in the tree view will update and display “123456 - HEX SCREW M8x20 A4-80 PASSIVATED” or “123456 - HEX SCREW M8x20 A4-80 PASSIVATED [3]” depending on what tab you’re watching (standalone part or assembly).

The text of the tree view cannot be edited directly for Part and Assembly containers. It is only possible to modify the two strings above. Bodies could be renamed manually as wanted. Features as well.

pierreporte, 2024-03-08:
@PaddleStroke Was my explanation sufficient?

PaddleStroke, 2024-03-11:
But this is still not an assembly issue. It's a core issue. And probably a tough one.

pierreporte, 2024-03-11:
> But this is still not an assembly wb issue. It's a core issue. And probably a tough one.

You’re right. Who may be able to give more details about how it could be done and how it would impact the ***@***.*** can you change the label of this issue and ping someone who

pierreporte, 2024-05-07:
#13859 would greatly benefit in having customizable labels in the tree view. I think that every entry of the tree view should use the formatted string instead of the raw names. The formatted string could of course contain only the raw name if necessary.

pierreporte, 2024-05-10:
Related forum thread: https://forum.freecad.org/viewtopic.php?t=87499

pierreporte, 2024-12-28:
On my 50-part assembly redesign to evaluate FreeCAD 1.0, I named Body (didn't use Part for convenience) and Assembly containers with the part number so that the assembly can be easily navigated. Besides issues mentioned in https://github.com/FreeCAD/FreeCAD/issues/12139#issuecomment-2564408323, when I forgot to rename the container and inserted it into the assembly, it was renamed Assembly001 as expected. However, renamind it afterwards doesn't change anything even with a recompute. This is because the names have no links with the children components. The proposal in this issue would solve the problem because it would fetch relevant information dynamically into linked files.

[image]

Ping @PaddleStroke