FreeCAD/FreeCAD #12142: [Problem] Assembly: isolating a set of parts is tedious

pierreporte, 2024-01-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working in large assembly, it is often needed to study how a few parts fit together. To do this, it is necessary to hide all the other parts, which is long, and to toggle their visibility back when finished. This is very long and impractical.

It would be useful if there was a feature to isolate selected parts. It would temporarily hide other parts with a command. A button somewhere would exit the isolation mode.

There could be an option to make the parts transparent instead of hidden.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-01-26:
There's a perfectly good macro that's been in existence for nearly three years which is the equivalent to Blender isolation mode:

```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
# Togglevis is a macro created for FreeCAD .19 and is intended to toggle the visibility of selected/non selected objects to make it easer to edit when you have many objects in scene
#
# HOW TO USE: select the objects you want to keep visible (any vertex, edge, face will work) then run the macro.
# All non selected objects will be hidden.
# Run the macro again to restore previsous visibility state.
#
__title__ = "ToggleVis.py"
__author__ = "Paulo Rogerio de Oliveira"
__version__ = "1.0"
__date__ = "2021/03/31"  # YYY/MM/DD

__Comment__ = "hide unselected objects so you can edit then easily. restore prior visibility state running the macro again"
__Help__ = "Select objects you must keep visible and run the macro. Run it again to restore previsous visibility"
__Status__ = "stable"
__Requires__ = "freecad 0.19 (not tested in previous versions)"

# import required libraries
import FreeCAD as App
import FreeCADGui as Gui
import PySide
from PySide import QtGui, QtCore
from PySide.QtGui import *
from PySide.QtCore import *

doc = App.ActiveDocument.Name
# clear output window
mw = Gui.getMainWindow()

# initialize cache
if not hasattr(FreeCAD, "previouslyVisibleObjects"):
    App.previouslyVisibleObjects = []
if not hasattr(FreeCAD, "previouslyHiddenObjects"):
    App.previouslyHiddenObjects = []

# store all parents of all selected objects to ensure its keep visible (store the Label)
# for parentList and selectedList, store the QTreeWidgetItem reference. Try to avoid error if change label while in edit mode
parentList = []
selectedList = []

# try to get internal object associated with QTreeWidgetItem
def getInternalObject(treeItem):
    # first looks by Label
    obj = App.ActiveDocument.getObjectsByLabel(treeItem.data(0, 0))
    if len(obj) > 0:
        return obj[0]
    else:
        # if not found, looks by Name
        return App.ActiveDocument.getObject(treeItem.data(0, 0))
    pass

# for each object in treeview
def traverse(list, pad):
    # for each child
    for idx in range(list.childCount()):
        item = list.child(idx)
        name = item.data(0, 0)
        obj = getInternalObject(item)

        isParent = name in parentList
        isSelected = name in selectedList

        # hide any object that is visible and not parent of selected or selected itself - store it on list to restore state later
        if obj.Visibility == True and not isParent and not isSelected:
            obj.Visibility = False
            App.previouslyVisibleObjects.append(item)

        # if not selected, not a group or if its parent of selected, traverse childs
        if not isSelected:
            if not hasattr(obj, "Group") or isParent:
                traverse(item, pad + "  ")

# if there are objects in visibility cache, you are in edit mode
inEdit = len(App.previouslyVisibleObjects) > 0

if inEdit:  # inEdit
    App.Console.PrintLog("Leaving Edit Mode\n")
    # show all objects that where visible when entered edit mode
    for object in App.previouslyVisibleObjects:
        obj = getInternalObject(object)
        try:
            obj.Visibility = True
        except:
            pass

    # hide all objects that where hidden when entered edit mode
    for object in App.previouslyHiddenObjects:
        obj = getInternalObject(object)
        try:
            obj.Visibility = True
        except:
            pass

    # reset visibility cache
    App.previouslyVisibleObjects = []
    App.previouslyHiddenObjects = []
    Gui.SendMsgToActiveView("ViewFit")
else:  # inEdit not
    tab = mw.findChild(QtGui.QTabWidget, "combiTab")
    tree = tab.widget(0).findChildren(QtGui.QTreeWidget)[0]
    top = tree.invisibleRootItem().child(0)

    # get selected objects
    sels = Gui.Selection.getSelection()
    if not sels:  # not sels
        App.Console.PrintWarning("Select something first!\n\n")
    else:  # sels
        for obj in sels:
            currentSelectedName = obj.Label
        App.Console.PrintLog(
            "Entering Edit Mode of " + doc + " - " + currentSelectedName + "\n"
        )
        # find the treeitem for ActiveDocument
        for idx in range(top.childCount()):
            doc = top.child(idx)
            if doc.data(0, 0) == App.ActiveDocument.Label:
                top = doc

        # store a list of selected items and its parents to keep it visible
        for sel in tree.selectedItems():
            current = sel
            selectedList.append(current.data(0, 0))
            while current:
                obj = getInternalObject(current)
                if hasattr(obj, "Visibility") and obj.Visibility == False:
                    App.previouslyHiddenObjects.append(current)
                current = current.parent()
                if current:
                    parentList.append(current.data(0, 0))

        traverse(top, "")
        Gui.SendMsgToActiveView("ViewFit")

```

maxwxyz, 2024-02-26:
Maybe related or can be combined with a kind of no-show space like CATIA: https://github.com/FreeCAD/FreeCAD/issues/11131

PaddleStroke, 2024-02-26:
You can use a the standard group object.
[image]
Does that answer the issue?

maxwxyz, 2024-02-26:
As I understand it, he only wants the currently selected part(s) visible. So select the object of interest and hide all the rest.

pierreporte, 2024-02-26:
Here is the workflow.

Say you have the following components in the tree view (can be at different levels but they aren’t in the example to simplify):

* Component 1
* Component 2
* Component 3
* Component 4
* Component 5 (hidden)
* Component 6 (hidden)

I want to isolate components 1 and 2 so I select them, then activate the command. The visibility is then:

* Component 1
* Component 2
* Component 3 (hidden)
* Component 4 (hidden)
* Component 5 (hidden)
* Component 6 (hidden)

When I’m done, I exit the isolation mode. The visibility is then back to what it was before:

* Component 1
* Component 2
* Component 3
* Component 4
* Component 5 (hidden)
* Component 6 (hidden)

The hidden space is an interesting idea but it is not what is discussed there.

maxwxyz, 2024-02-28:
Maybe a solution would be not only helpful for the Assembly WB but a function generally available in Core / different WB.
Solution in CATIA is, to only show the currently active body / parts. So you could select and activate an object and only this is visible. Select and activate another one and only this one is visible. All other bodies / parts are hidden, which are not activated.

Described in the bottom half here: http://catiadoc.free.fr/online/bascuprt_C2/bascuprt0100.htm
Demo:
https://youtu.be/ohwcoRYms5Y?feature=shared&t=278

[image]

pierreporte, 2024-02-28:
@maxwxyz indeed definitely not for the assembly workbench, if only because you can open components in a separate tab. The Catia feature is great but it needs a good visual clue that it is activated. At my job we call it the “idiot mode” because you feel like an idiot after realizing that your bodies didn’t just vanished overnight.

A good visual clue would be to still show other bodies but very transparent. In fact it would be useful for isolate mode as well, with a toggle to switch from the true isolate mode and the slightly less isolated mode (which doesn’t allow to pick transparent parts).

pierreporte, 2024-02-28:
I propose this for the UI. Once isolate mode (or current body mode) is activated, there is a button near the navigation cube and above the task panel. Clicking it restores the view. I put it there because it’s near the view menu that the cube is becoming, it doesn’t need to move when the panel is visible, and it doesn’t mess with the tree view.

[image]

Solidworks does something similar with a floating window:

[image]

drwho495, 2024-05-15:
I am planning on adding an isolate feature to my fork of freecad, so it can get merged here. I can fill you in on progress as I go.

pierreporte, 2024-05-15:
@drwho495 Great! It could land in 1.0 if you managed to make your PR ready for review in less than two weeks.

drwho495, 2024-05-22:
[image]
here is the progress, i need to do some work on the backend for some failsafe and a tool to fix any "stuck" transparent objects.

pierreporte, 2024-05-22:
It’s a bit hard to understand with the animated picture. Is it necessary to select in the task panel what parts to isolate? Are the transparent parts pickable? They shouldn’t be pickable.

drwho495, 2024-05-22:
So you can use the task panel if you don't select a part, but if you do select a part the task panel will not appear because it is not needed

pierreporte, 2024-05-23:
I mean, are the parts selectable when the opacity is not 100%?

drwho495, 2024-05-23:
> I mean, are the parts selectable when the opacity is not 100%?

Yes, but I believe that there is no way to make them passthrough in that way

drwho495, 2024-05-23:
> Then I think that it would be best for the first implementation to avoid supporting transparency. The transparent unselectable parts are useful for more, like contextual editing of parts.

I did it like this to because I model all of my features off of onshape, and I actually improved onshape's because you can always select the hidden parts even if fully transparent. Contextual editing of parts is also impossible from directly within the assembly wb (because of cyclic dependency errors), that is why I created my context creator tool because it enables for this feature to be possible without any cyclic problems in any way

pierreporte, 2024-05-26:
I took the time to better think about this and to make a mockup for the GUI.

First, the isolate mode shouldn’t be limited to parts and sub-assemblies. Sketches, datum, and basically everything in the design tree should be selected. I don’t think that the task panel is necessary because it would duplicate the tree view.

Second, when in isolate mode, the hidden objects should have the eye icon set accordingly, and switching it back open brings the objects in the visible space (like in Solidworks).

Last, the GUI. I propose a very small window, always on top, that cannot be closed, similarly to Solidworks. It would be in one corner of the 3D view, freely movable by the user. It would be limited to one button to exit the mode, two radio buttons to hide the other objects (selected by default) or make them transparent, a checkbox to allow selection of non isolated objects (unchecked by default). The opacity settings would be better in the program settings, that could let the user determine if other objects should be hidden or transparent by default, as well as selectable or not.

[image]

@FreeCAD/design-working-group Any idea for UI/UX?

@yorikvanhavre Would this feature be useful for architecture? If so, it would be better to implement it in a more generic place.

PaddleStroke, 2024-05-28:
I liked the transparency slider. If you want visible but not transparent you set it to 100. If you want hidden you set it to 0. And its more flexible than radio button because it let you choose the transparency.

Is there a reason to use a qdialog rather than the task panel? Is it because you want to be able to do other tasks while in isolate mode? In which case it makes sense to us qdialog as to let the task panel free.

drwho495, 2024-05-28:
> I liked the transparency slider. If you want visible but not transparent you set it to 100. If you want hidden you set it to 0. And its more flexible than radio button because it let you choose the [...]

Yes, it makes sense to me that you would want to, let's say, edit a part quickly in the assembly wb without switching files. There actually have been times where I h ave needed this tool.

kadet1090, 2024-05-28:
> Last, the GUI. I propose a very small window, always on top, that cannot be closed, similarly to Solidworks. It would be in one corner of the 3D view, freely movable by the user. It

While I like this approach we should stick with the task panel for now and not introduce new UI constructs for now. This is however something that could be addressed as bigger overhaul of task panel UI that would also affect sketcher tool settings (OVP).

kadet1090, 2024-05-28:
I did not say that it is not helpful, nor that it is bad, conceptually I fully agree with that. I simply meant that it should be implemented in a way that integrates well with rest of FreeCAD and can be used by other workbenches for other tasks.

This however require more work to be done and should take into account all possible scenarios that could use such windows. That's why for this PR we should stick with what we already have and implement better UI solution later.

drwho495, 2024-05-28:
We could have a radio for a "transparent mode" and have the slider become unusable. But I think the more minimalist idea of the slider makes sense because it combines all of the features we need into one while being easy to use and not take up space.

pierreporte, 2024-05-28:
Here is an idea for the slider. It is based on sound level selectors commonly found in operating systems. At one side there is 0 % opacity, at the other side 100 % opacity. Somewhere in the middle, there is a magnetic tick for the preferred transparency level (the value is a program settings). Below is an illustration. Though this is likely taking more space than a select box in any case.

[image]

pierreporte, 2024-05-30:
@drwho495 FYI the feature freeze is Monday 3rd June. If you want this feature in the 1.0 version, you must have a ready for review PR by that time.

pierreporte, 2025-02-17:
Interesting UI proposal that could work: https://github.com/FreeCAD/FreeCAD/issues/5563#issuecomment-2664025608

drwho495, 2025-03-12:
I'm going to get my isolate tool working, then I can PR it into main. I'll link videos of it working when I'm done.

pierreporte, 2025-03-12:
Will you use the UI proposal linked above?

drwho495, 2025-03-12:
the github link is https://github.com/drwho495/FreeCAD on the main branch. Feel free to make any changes to the ui and send it to me or PR it into the repo.

maxwxyz, 2025-03-12:
What happens to the object outside the assembly container, are they also being changed? "Done" applies the transparency or resets it?

drwho495, 2025-03-12:
> What happens to the object outside the assembly container, are they also being changed? "Done" applies the transparency or resets it?

It references all of the objects in the assembly and it resets it 100 for right now, I plan on setting two temporary attributes, one called original transparency, and another called isIsolated so if the program crashes when isolated it can be reset back to normal with another command. The attributes will be deleted after you close the isolate menu or reset everything.

maxwxyz, 2026-11-07:
This is available in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/56b02196a927ca7fef7bee9804d55cbd30f71128

Isolate Selection keeps the selected parts pickable and makes the others transparent, wireframe or hidden. End Isolation restores the previous state.

PaddleStroke, 2026-11-07:
@maxwxyz please note that these commits are unreviewed AI generated. Which is why I do not PR them. It's made with opus 5.5 which I trust to do right most of the time, but it is likely not the best solutions.
So these commits can be used by anyone wanting to tackle this project as reference but not blind cherry picking.

PaddleStroke, 2026-11-07:
Also I do not know if you have seen, but these issues appear in the ticket already : 

[image]