FreeCAD/FreeCAD #12144: [Problem] TechDraw.writeDXFPage export only available in Gui module, not in headless

3xxx, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

in GUI use the python code below can export dxf file correctly,but in headless,the code below can't export a dxf file correctly.
```
>>> ### Begin command TechDraw_ExportPageDXF
>>> import TechDraw
>>> TechDraw.writeDXFPage(App.getDocument('test9').getObject('Pagina1'),u'C:/Users/Administrator/Downloads/Pagina di prova 1.dxf')
>>> App.ActiveDocument.recompute()
>>> ### End command TechDraw_ExportPageDXF
```
[1.zip](https://github.com/FreeCAD/FreeCAD/files/14071829/1.zip)

### Full version info

```shell
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35510 (Git)
Build type: Release
Branch: main
Hash: da05f7c8e8b25ff783cc6c4eb3b73b640f134519
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.60h
  * Assembly3 0.11.4
  * Assembly4 0.12.5
```
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-01-27:
This looks like a duplicate of #5710.

3xxx, 2024-01-28:
Hi [WandererFan](https://github.com/WandererFan), Thanks for your answer.

closed 2024-01-28 by 3xxx