FreeCAD/FreeCAD #12145: [Feature Request] Sketcher: show dimensions out of edit mode

FlachyJoe, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We currently can't see sketches dimensions without open them in edit mode. A specific view mode could be created in order to solve this.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35726 +1 (Git)
Build type: Release
Branch: short_suppress
Hash: c61705227a9b8c149bcf33e665846ddc9dc79eb0
Python 3.11.7, Qt 5.15.10, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-01-27:
Do you also want the user to be able to edit the dimensions from outside the sketch?

FlachyJoe, 2024-01-27:
@pierreporte no, I think it's not a good idea.

maxwxyz, 2024-03-06:
I think it would be, maybe there could be a central hub to edit (named) dimensions in a document. Maybe something connected to https://github.com/FreeCAD/FreeCAD/issues/12120
And I've created a related one to this: #12795

FlachyJoe, 2024-03-06:
> maybe there could be a central hub to edit (named) dimensions in a document.

I've in mind a Expression workbench to list them all and help to create new ones.
It's quite easy to do in python by parsing ExpressionEngine of all objects, it almost only needs a GUI.
```python
>>> doc = App.ActiveDocument
>>> [(o.Label,o.ExpressionEngine) for o in doc.Objects if o.ExpressionEngine]
[('Sketch001', [('Constraints[10]', '.Constraints.Length')]), ('Pad', [('Length', '5 * Sketch001.Constraints.Length')])]
```
Your request could take place in a such WB.