FreeCAD/FreeCAD #12147: [Problem] Cannot open an addons manager

zamonary1, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After clicking on addon manager button my manager doesn't open and just hangs on the "checking the connection to github" window. In the console there is one error that indicates of some kind of problem with QT.
[image]
Some time ago that was working fine but then it somehow stopped working. Maybe i have just deleted some crucial dependencies with apt?

### Full version info

```shell
[code]
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Curves 0.6.13
  * LCInterlocking
  * sheetmetal 0.3.1
  * CadbaseLibrary 0.3.0
  * retr3d
  * 3D_Printing_Tools
  * fasteners 0.4.66
  * InventorLoader 1.4.0
  * A2plus 0.4.61
  * freecad.gears 1.0.0
  * Design456 0.0.1
  * AnimationFreeCAD 1.0.0
  * CurvedShapes 1.0.5
[/code]
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-02-26:
I believe I have resolved this issue: @zamonary1 can you test it out on the latest Weekly and see if it's working now?

zamonary1, 2024-03-01:
> I believe I have resolved this issue: @zamonary1 can you test it out on the latest Weekly and see if it's working now?

Thank you for replying. Tried the latest weekly build and when i try to open addon manager i get the same window "Checking for connection to github..." but nothing happens next. The error from the console is gone, though. I may try using internet packet analysis or buy a vpn.
Also, weekly build looks amazing! When i switch views it interpolates the animation and it looks way smoother than stable. The draft workbench's auto constraints when making new objects are very handy too. Can't wait to try it out fully!

luzpaz, 2024-03-09:
@zamonary1 can you test again on latest dev version and post the full [About](https://wiki.freecad.org/About) info? TIA

JJTechman68, 2024-04-29:
I am having the same issue on Ubuntu 20.04LTS and the latest dev build

OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git) AppImage
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Installed mods: 
  * boltsfc 2022.11.5
  * ThreadProfile 1.89.0
  * CurvedShapes 1.0.9
  * toSketch 1.0.1
  * Silk 0.1.5
  * FeedsAndSpeeds 0.5.0
  * MeshRemodel 1.9.0
  * sheetmetal 0.3.12
  * parts_library
  * Defeaturing 1.2.2
  * Render 2023.8.18
  * MakerWorkbench 1.0.1
  * lattice2 1.0.0
  * ExplodedAssembly
  * Pyramids-and-Polyhedrons
  * dodo 1.0.1
  * ose-piping
  * LCInterlocking
  * fasteners 0.4.71
  * symbols_library
  * woodworking 0.21.2.33771
  * A2plus 0.4.64a
  * BOLTSFC
  * InventorLoader 1.4.0
  * QuickMeasure 2022.10.28
  * Assembly3 0.12.2
  * Manipulator 1.5.7
  * Curves 0.6.35
  * SlopedPlanesMacro
  * kicadStepUpMod 10.22.1
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * Assembly4 0.50.12
  * workfeature