FreeCAD/FreeCAD #12148: [Problem] Path: Creating a opensbp gcode file generates error

kzembower, 2024-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating an OpenSBP ShopBot gcode file generates this error:
```
07:28:27  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "/tmp/.mount_freecatszyYk/usr/Mod/Path/Path/Post/Command.py", line 590, in Activated
    finalgcode += gcode

can only concatenate str (not "NoneType") to str
```
However, the file is created and seems to work correctly (haven't run on ShopBot yet).

-Kevin

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.62
  * FeedsAndSpeeds 0.5.0
  * Assembly3 0.12.1 (Disabled)
  * Help 1.0.3
  * Assembly4 0.50.6
  * woodworking 0.21.1.33694
[/code]
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-03-13:
@kzembower can you reproduce on latest dev version ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

kzembower, 2024-03-18:
@luzpaz, sorry, don't have time to test on dev version at this time.

Steps to reproduce this issue: generate an OpenSBP ShopBot gcode file.

Sorry I can't be of more help.

-Kevin

luzpaz, 2024-07-13:
Hi @jkvfc, I've noticed you've been active in some of the CAM related ticked on the bug tracker. Would you have a moment to confirm this issue ? TIA

jkvfc, 2024-07-14:
Hi @luzpaz, I'm able to confirm this issue. In my FreeCAD version (see below) the error message looks a bit different.

[image]
My test File
[TestOpenSBP.zip](https://github.com/user-attachments/files/16216023/TestOpenSBP.zip)
I don't know anything about Open ShopBot. I hope this still helps.

OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38098 (Git)
Build type: Unknown
Branch: main
Hash: 5b9b611f8ec223a8bc3c8739fee92bea3b4a5705
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)

LarryWoestman, 2025-06-10:
I just tried to reproduce this problem on a very current build using the TestOpenSBP.FCStd file and one other file and everything appeared to work normally for me.

I know that many of the postprocessors including the opensbp postprocessor have had some recent changes related to code related to writing out files.  I suspect that the recent changes may have fixed this issue, but have no way to prove that.

Unless someone can find a way to reproduce this issue on the current dev version of FreeCAD, I suggest that this issue be closed as "not reproducible".

LarryWoestman, 2025-06-10:
(Not reproducible any more, anyway :-)

luzpaz, 2025-07-13:
@kzembower we're closing this ticket. If you can still reproduce on latest 1.1-dev then please re-open.

closed 2025-07-13 by luzpaz